gotthardp/lorawan-server #145: Kerlink Lora 868 IoT Station connection to LoRa server problem

Gligor721, 2017-07-06:
Hello, 

I use this LoRa server with microchip base LoRa gateway and all is working well but now I switch to Kerlink Lora 868 IoT Station. I power up the gateway and install Autorun packet forwarder on the gateway. I configure the packet forwarder server ip and the port but on the server debug log i not see that the gateway is connected to server! Is this server test with Kerlink Lora 868 IoT Station?Can you help me so I can make this working?Is there any configuration on the server I need to do for this?
I`m connected to the gateway via ssh so I see that the kernel and Linux is up and ruining and all is ok but no communication between gateway and server!

Thanks for all your help!

gotthardp, 2017-07-06:
Hello. Kerlink is not tested yet, but as long as it runs a standard packet_forwarder it shall work. Did you have an option to set the server port or not? Sometimes gateways use port 1700 as a default, but server listens at 1680.

Gligor721, 2017-07-06:
In the gateway configuration I have json config file where I can configure server ip and the ports.Port is the same on the server and gateway and the IP is configure correctly.
Json file example:
"gateway_conf": {
    "gateway_ID": "0000024b080503a4",
    "serv_port_up": 1680,
    "serv_port_down": 1680,
    "server_address": "iot.semtech.com" here i change to xxx.xxx.xxx.xxx
}

gotthardp, 2017-07-06:
Alright, you will need to do some more debugging then. Could you please check if the UDP packets from the gateway reached the machine hosting the server and eventually the server itself? Do you see a "gateway: connected" message in the logs and in the server admin? I assume no errors are printed in neither the gw nor the server log, true?

Gligor721, 2017-07-06:
There is no "gateway: connected" message in logs!I need to write a udp server and start on my machine so I test if the gateway send udp packet to my machine?

gotthardp, 2017-07-06:
No. You could run `tcpdump -i eth0 -w file.pcap` to capture packets and then run a Wireshark to find any on port 1680. You could also stop the server and run `nc -ul 1680` to have a simple UDP server receiving data from the gateway.

Gligor721, 2017-07-06:
Nothing received from gateway when i tun nc -ul 1700 nothing is received.I thing that the gateway is not sending nothing but i don`t know why.Can you help me to debug this problem?

gotthardp, 2017-07-06:
you should do `nc -ul 1680`

Gligor721, 2017-07-06:
I change port on the server  microchip gateway is sending on 1700 port so i change and is working.
Now port is change on the kerlink gateway to 1700.

gotthardp, 2017-07-06:
So, does it work, or you need some further assistance?

Gligor721, 2017-07-06:
It is not working I test with wireshark and server don`t received any udp message from gateway!
I think that the problem is in the gateway but i don`t know how to solved this. My microchip nodes not view any gateway.

gotthardp, 2017-07-06:
Mmm. Unfortunately, I have zero experiences with Kerlink, so I am not sure what to do either. (I have one Kerlink gateway in the lab, which I am about to test soon, so I may gain some experiences in the near future.)

Gligor721, 2017-07-07:
Info: server now is communicating with kerlink gateway!Now the problem is that the gateway is not receiving all packet, I received 4 packet and now not receiving.Can you have idea what is the problem?

Gligor721, 2017-07-07:
Problem is solved!Server is working with KERLINK very good!
Thanks for all your help.
p.s. I will install the lasts server now, i tested with 0.4.1.
Is there a big update in lasts server version?

gotthardp, 2017-07-07:
Cool! Could you please explain a bit what was the issue and how you solved it? Others may face the same problem.

Since 0.4.0 I do mainly bug fixes and minor enhancements (which are summarized in the release notes). The update is not big, but may be significant depending on what features you use.

closed 2017-07-15 by gotthardp