gotthardp/lorawan-server #147: Deduplication of gateways with long delay

gotthardp, 2017-07-06:
When we have gateways with long delays (3G connected):
 * the de-duplication timer may fire before data from all gateways are received (we see two frames in the list)
 * this unnecessarily delays the response

Refactoring of this is required to:
 * not waste time waiting for other requests (can start processing immediately)
 * decide "best gateway" just before the response is about to send

For debugging purposes log the reception from all gateways, but **after** processing the data.

UzixLS, 2017-07-13:
Can this make OTAA impossible in cases when device in scope of multiple gateways with significantly different pings?

gotthardp, 2017-07-13:
I don't think so. The join request from the first (fastest) gateway will be relayed to the server and processed and then the join response will be sent through the gateway that received the best signal quality.

gotthardp, 2017-12-28:
done in the latest devel branch

closed 2017-12-28 by gotthardp