gotthardp/lorawan-server #148: HTTP Backend Connectors

Quentintin, 2017-07-06:
Hi,

I would like to know if you plan to add HTTP/HTTPS Backend connectors ? 
It will be awesome if you can provide it. 👍 

Thanks for response :)

gotthardp, 2017-07-06:
Simple connector making one HTTP/S request per frame, without support for downlinks is quite easy. The reason why it's not implemented is that I was not sure if anyone needs such simple thing, because there may be a lot of variations (HTTP GET vs. HTTP POST and various content types).

What are your needs? Where are you connecting to?

Quentintin, 2017-07-06:
I created backend to receive messages from Sigfox and LoRaWAN Operators. Most of them provides callback methods with HTTP/S. 
My backend is a part of a Web Platform to manage IoT devices. I implemented HTTP restAPI used by users and Operator's backend to push messages. That's why I need this.

If it simple for you, you should implement it because i'm pretty sure lot of people will ask you for that connector.

gotthardp, 2017-07-06:
So far you are the first asking for this :) but I can do that. Do you have some examples how this HTTP API often looks like? What URI, Content-Type, Body?

Quentintin, 2017-07-06:
The perfect callback for me is POST method :
- URL (can content variables {devEUI}, {devAddr}, {...})
- Content-type : application/x-www-form-urlencoded or application/json
- Headers : For authentication - Authorization
- Body : Editable fields with variable, or static content

Firstly you can do easy way with static url/body, and later allow custom fields.

Sorry for my english :/ And thank you again

gotthardp, 2017-07-06:
Will do. :-)

stephanedavid, 2017-07-19:
Some news about this topic, we are also very interested by this connector ?

gotthardp, 2017-07-19:
Not yet, but the more people are interested the higher priority it gets. ;-)

Is your view on a perfect HTTP callback the same as described above, or you have some other needs?

stephanedavid, 2017-07-19:
Exactly the same view.
Thanks

mkpuig, 2017-07-21:
Hello, i'm interested too.
Thanks a lot for your great work!

gotthardp, 2017-07-23:
@Quentintin @stephanedavid @mkpuig I started to work on this. In the 'master' branch there is an experimental implementation already available (which you can try): you can create an HTTP(s) connector and define some "Published Topic". The content-type is determined based on the handler format.

If you encounter some error, just let me know.

I am still not satisfied with the error handling, so in case an error occurs the system may crash in a not so nice way. This will be fixed in the coming week(s).

Kralg, 2017-07-31:
Hello, I would like to test the server version with REST commands for txframes. Can you explain on how can I do that? Should I download files from test folder or somewhere else? I would also like to know when it is possible to expect the 0.4.8 version? I would be really great to get it.

gotthardp, 2017-07-31:
Hello. I want to do some more testing to make sure I didn't destroy anything, but hope to release 0.4.8 today,

gotthardp, 2017-07-31:
v0.4.8 released
Feel free to open a new issue if something goes wrong.

closed 2017-07-31 by gotthardp