gotthardp/lorawan-server #151: Lorank8 gateway issue

kazi93, 2017-07-07:
Currently two servers IMST and Lorank8 are connected with the server in a local network. However, I am receiving the below warning though both gateways are sending frames to the server. 

22:15:51.738 [warning] Unknown element in JSON: {prf,#{dw_bcn => 0,dw_rf => [1,0],dw_srv_dg => [1,0],dw_srv_hb => [1,0],up_rf => 0.5987940103,up_srv_dg => [1,0]}}
[warning] [Client <0.654.0>] Connection lost for: tcp_closed
22:15:54.744 [debug] Connector attempt31 {shutdown,tcp_closed} reconnect 63
22:16:00.205 [debug] DataRate <<38,1,27,100>>: average snr 8 2.5 = 5, dr 5 -> step 1
22:16:00.206 [warning] MQTT broker disconnected, message lost
22:16:21.755 [warning] Unknown element in JSON: {dev,#{alt => 0,desc => <<>>,email => <<>>,gps => <<"fake">>,id => <<"1dee02de5c47de9b">>,lat => 0,lon => 0,pfrm => <<"Lorank">>,up => 9017}}
22:16:21.755 [warning] Unknown element in JSON: {prf,#{dw_bcn => 0,dw_rf => [1,0],dw_srv_dg => [1,0],dw_srv_hb => [1,0],up_rf => 0.7993970052,up_srv_dg => [1,0]}}
22:16:35.966 [debug] DataRate <<38,1,27,100>>: average snr 8 2.5 = 5, dr 5 -> step 1
[warning] [Client <0.657.0>] Connection lost for: tcp_closed
22:16:39.749 [debug] Connector attempt311 {shutdown,tcp_closed} reconnect 64
22:16:51.761 [warning] Unknown element in JSON: {dev,#{alt => 0,desc => <<>>,email => <<>>,gps => <<"fake">>,id => <<"1dee02de5c47de9b">>,lat => 0,lon => 0,pfrm => <<"Lorank">>,up => 9047}}
22:16:51.761 [warning] Unknown element in JSON: {prf,#{dw_bcn => 0,dw_rf => [1,0],dw_srv_dg => [1,0],dw_srv_hb => [1,0],up_rf => 0.8996985026,up_srv_dg => [1,0]}}

gotthardp, 2017-07-07:
Lorank8 strikes again :) The problem is their proprietary message format. In your gateway config you should have "stats format" set to "ideetron concise" (or so). When you set this to "semtech" you should get this work.

If you succeed I will document this somewhere.

gotthardp, 2017-07-15:
I just documented that in the Troubleshooting guide.

closed 2017-07-15 by gotthardp