gotthardp/lorawan-server #155: No confirmation (ACK) sent to the node

LucaBertuolo, 2017-07-12:
Hi
I'm trying your great lorawa-server, but i'm having an issue with a confirmed packet sent from the node...

The node is a Waspmote + RN2483 in ABP mode, in the Nodes tab i've add a node with this setting:
https://s17.postimg.org/l7mxzh00f/Schermata_da_2017-07-12_11-36-27.png

and in the note i've load an example code that send a Confirmed packet in ABP mode every 20sec, if it don't receive the confirmation ACK, it will try 4 times...
It always try 4 times because never receive the ACK from the gateway, in fact i can also see that in the "Received frames" tab it never tick the "confirmed"...
https://s13.postimg.org/t0rxx5ztj/Schermata_da_2017-07-12_11-42-19.png

I'm missing some settings?
In  TX Chain I've correctly set the right channel: 0
in TX Window i've try wit 1 and 2, but nothing change...

gotthardp, 2017-07-12:
Hi. I cannot say more that it's written in the https://github.com/gotthardp/lorawan-server/blob/master/doc/Troubleshooting.md#no-downlink-frames-delivered

If neither the server nor the gateway report any error it's a device/gateway issue. It's probably the hardest problem to debug, but it's impossible for me to help.

gotthardp, 2017-07-12:
You will not receive an ACK if the message you sent does not request Confirmation. You should configure your device to send a Confirmed uplink.

LucaBertuolo, 2017-07-13:
That's very strange, I've do this test:

1) I've load an example sketch in my waspmote/arduino node for send a Confirmed package
2) I've run the lora_pkt_frwd utility, and listen in the UDP port for a message

In the UDP Upstream port i can correctly get my packet: gAcABgUABAAD0zLXeLcl7Q==
and decode it online:
https://lorawan-packet-decoder-0ta6puiniaut.runkit.sh/?data=gAcABgUABAAD0zLXeLcl7Q%3D%3D

But if i stop the lora_pkt_frwd and i change the ip server to send the udp packet to your server on my pc, i cannot find the packet in "received frame"
But if i use a unconfirmed sketch in my node, i can correctly find the packet in "received frame"

gotthardp, 2017-07-15:
Is there some error or warning in the server log?

gotthardp, 2017-07-21:
Closing this for now. Feel free to re-open the issue when you manage to get more info on this issue.

closed 2017-07-21 by gotthardp