gotthardp/lorawan-server #156: Re-joing bugginess

SayreBlades, 2017-07-12:
We are testing a scenario whereby our sensor is re-joining the network, but prior to the re-join the sensor's appeui has been changed.  We suspected this should be rejected by LNS but it is not.

Even after deletion of the original device and node within LNS, and then re-creating the device with original appeui/deveui, the sensor rejoin is successful.  This seems to be invalid behavior.

Here is a screen shot if that helps (sensitive information F'd out).  We are using OTAA.
[image]

gotthardp, 2017-07-12:
The AppEUI is not used in the server. Shall it be?

SayreBlades, 2017-07-12:
FYI, it seems this behavior isnt actually particular to a re-join scenario.  On the first join our sensor was able to successfully join with inconsistent appeui's.

SayreBlades, 2017-07-12:
Hi Gotthard, Im actually not sure if it should be.  Let us review the lorawan spec.  It just seems to be incorrect behavior, but it may actually not be specified.

gotthardp, 2017-07-12:
Yeah. It's never checked, because it's just an ID that is useless in case of this server, but let's consider this a bug. I will:
 - make AppEUI optional, so it can be disabled if someone doesn't like it
 - check the AppEUI against what device sent and issue an error if they don't match

gotthardp, 2017-07-12:
I just implemented that and will include that in next release. Please check.

SayreBlades, 2017-07-13:
Thanks much!

closed 2017-07-15 by gotthardp