gotthardp/lorawan-server #157: I  Can't view API Administration page

suhaibfs, 2017-07-13:
Hello there,
I am new to LoRa, I installed Erlang 20 on my linux server, then I installed lorawan-server using .deb package. I allowed 8080 port on my firewall, yet I can't view the API page.
Note : I am not running the server locally.

What can be the issue here?

Thanks in advance!
[image]
[image]

gotthardp, 2017-07-13:
And how do you access the API page? What error are you seeing?

suhaibfs, 2017-07-13:
I am supposed to open http://MYSERVERIP:8080 to access it right?
when I open that in my browser I get time out error
This site can’t be reached
[Server IP] took too long to respond.

gotthardp, 2017-07-14:
That is weird. I see that 80 and 443 is open too, could there be some httpd/proxy related issue?

suhaibfs, 2017-07-15:
The issue was from the server provider side (Alibaba Cloud), they have ( Security Group List ) where I had to allow 8080 PORT access. Now everything works fine ( server side ), moving to setting up my Raspberry PI for packet forwarding.
Thank you for your help.

gotthardp, 2017-07-15:
Cool. Thanks for letting me know.

closed 2017-07-15 by gotthardp