gotthardp/lorawan-server #158: problem with installation

pdl86, 2017-07-13:
While trying Lorawan-server on 
windows 10 / Erlan 19 or 20 

a) the  location of the "lorawan-server.bat" file is incorrectly described in the documentation as "bin" while it is  "scripts" 
b) when launching the bat file, it failed  stating 

"could not start kernel pid (application_controller) (error in config file "releases/{{release_version}}/sys.config" (none): configuration file not found)

Crash dump is being written to: erl_crash.dump...done
"

there is no "sys.config" file in erl realeases/release_version directory (  both 19 and 20 ) 

( unless the erl install is different on Linux, as the command is almost the same in the bash script, it may fail also on linux)

c) there is an issue with the command  "set FILES=!FILES! %%A\ebin", when %%A contains a space 
( strange as when used in the third command of the script , it works )

Any help will be appreciated 
best regards
philippe

gotthardp, 2017-07-13:
Hello. The "lorawan-server.bat" is actually in the "bin" directory. You probably downloaded a source package (where the .bat is in scripts), which cannot be executed. Please use the binary release.

gotthardp, 2017-07-13:
the "set FILES=!FILES! %%A\ebin" related issue is a bug, which I will fix

gotthardp, 2017-07-15:
I just fixed the space-related issue. Thanks for reporting this.

pdl86, 2017-07-17:
Hi Petr

I have nstalled  the 0.4.7, this time as a binary,  ( oops ) , but the "banl" issue is still there 
if install in a directory as "D:\No Backup\temp-install\lorawan-srv-gotthardp\lorawan-server\bin"
the results in the FILES variavle is still"No"
if installed in "D:\test\lorawan-sver\bin" the FILES variables is correctly set 

so there is a trick somwhere, and I'm not sufivently skille din the cmd subtle things to find it .

pdl86, 2017-07-17:
@gotthardp

pdl86, 2017-07-17:
@gotthardp 
please have a look, the "bug" on blank intoa directory name is still there

gotthardp, 2017-07-21:
@pdl86 I just made another fix. You can have a look at the above mentioned patch and move the `"` quotation marks are indicated, or wait for the next release (in 1-2 weeks). Thanks for reporting this.

gotthardp, 2017-07-31:
V0.4.8 released

closed 2017-07-31 by gotthardp

pdl86, 2017-09-13:
Hi

thanks

confirm it works.

regards

philippe

olegscorpion, 2022-08-03:
Hello Gotthardp name is Oleg. i'm from Kazakhstan.
i need help.
I am trying your server. Installed Earl 24, after starting the bin server.bat gives an error:
C:\lorawan-server-0.6.7\bin\lorawan-server.bat
Server started at http://localhost:8080
{"init terminating in do_boot",{undef,[{lorawan_app,start,[],[]},{init,start_em,1,[{file,"init.erl"},{line,1220}]},{init,do_boot,3,[{file,"init.erl"},{line,910}]}]}}
init terminating in do_boot ({undef,[{lorawan_app,start,[],[]},{init,start_em,1,[{_},{_}]},{init,do_boot,3,[{_},{_}]}]})

Crash dump is being written to: erl_crash.dump...done

Windows 7 OS. What did I do wrong?

olegscorpion, 2022-08-03:
i install earl 19.2 ? it all working.