gotthardp/lorawan-server #160: installation on Raspberry Pi 2 Latest Jessie Fails.

Rustie0125, 2017-07-14:
Non of the menthods used in the installation guides works correctly. fresh jessie install run the package as follows but nothing 
Get the latest sources by:

git clone https://github.com/gotthardp/lorawan-server.git
cd lorawan-server
If you already obtained the sources you can upgrade to the latest version by:

cd lorawan-server
git pull
make upgrade
Then build and release the lorawan-server by:

make release

does not succesfully upgrade or release

gotthardp, 2017-07-14:
You mean new installation or upgrade? Upgrade is broken I know.

Rustie0125, 2017-07-15:
New install 

Please safe our environment and do not print this email.

Rustie0125, 2017-07-15:
It might be something im doing wrong but i have tried it a few times and get different results. the installation and running for Linux instructions is not very clear.

gotthardp, 2017-07-15:
What errors are you getting from the `apt-get install`?

Rustie0125, 2017-07-18:
Following the instructions from the top this is the result to the first command 
root@raspberrypi:~#  lorawan-server-0.4.6.deb
-bash: lorawan-server-0.4.6.deb: command not found
root@raspberrypi:~# root@raspberrypi:~#
-bash: root@raspberrypi:~#: command not found
root@raspberrypi:~#

When following the the steps for linux binary 
this is result. 

`
lib/worker_pool-2.2.3/ebin/wpool_time_checker.beam
lib/worker_pool-2.2.3/ebin/wpool_utils.beam
lib/worker_pool-2.2.3/ebin/wpool_worker.beam
releases/0.4.6/vm.args
root@raspberrypi:~/lorawan-server# bin/lorawan-server
bin/lorawan-server: 24: bin/lorawan-server: erl: not found
root@raspberrypi:~/lorawan-server#
`

Rustie0125, 2017-07-18:
I reloaded the image onto the PI and started again. When i run the command to start the server i get the following result.

` root@raspberrypi:~/lorawan-server# bin/lorawan-server
-bash: root@raspberrypi:~/lorawan-server#: No such file or directory
root@raspberrypi:~/lorawan-server# {"init terminating in do_boot",{{badmatch,{error,{lorawan_server,{bad_return,{{lorawan_app,start,[normal,[]]},{'EXIT',{{badmatch,{error,prerequisite_failed}},[{lorawan_app,start,2,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_app.erl"},{line,19}]},{application_master,start_it_old,4,[{file,"application_master.erl"},{line,272}]}]}}}}}}},[{lorawan_app,start,0,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_app.erl"},{line,16}]},{init,start_it,1,[]},{init,start_em,1,[]}]}}
-bash: init terminating in do_boot: command not found
root@raspberrypi:~/lorawan-server#
root@raspberrypi:~/lorawan-server# Crash dump was written to: erl_crash.dump
-bash: Crash: command not found
root@raspberrypi:~/lorawan-server# init terminating in do_boot ()
-bash: syntax error near unexpected token `('
root@raspberrypi:~/lorawan-server# root@raspberrypi:~/lorawan-server#
-bash: root@raspberrypi:~/lorawan-server#: No such file or directory
root@raspberrypi:~/lorawan-server#
`

gotthardp, 2017-07-18:
First of all, if you are using the .deb image, install it using `sudo dpkg -i /path/to/deb/file` and start it by `systemctl start lorawan-server`. The `bin/lorawan-server` command is intended for those who are not using the .deb package.

Rustie0125, 2017-07-18:
Hi Petr

Thank you for the reply, I have tried both mentioned neither works. 

Please safe our environment and do not print this email.

gotthardp, 2017-07-18:
If you share with me with the details (the error logs) I may be able to help you.

Rustie0125, 2017-07-18:
I tried to find the erl_dump file reported in the log but could not find it
manually or with file search via winSCP

gotthardp, 2017-07-18:
Could you please:
 * Start with a fresh PI image
 * Get the latest .deb
 * Install it using `sudo dpkg -i /path/to/deb/file`
 * Start it using `systemctl start lorawan-server`
 * Let me know how it went?

Do NOT try anything else, otherwise you mess up the logs and everything.

Rustie0125, 2017-07-19:
following the steps above seems to produce no errors, but when i try open the webpage on any of the following it does not work. 
192.168.11.17:8080
192.168.11.17:1680

gotthardp, 2017-07-19:
Please try `systemctl status lorawan-server` to get some info on the service

Rustie0125, 2017-07-19:
in the log files same error is reported as before.

> 2017-07-19 04:46:39.693 [error] <0.120.0> CRASH REPORT Process <0.120.0> with 0 neighbours exited with reason: no match of right hand value {error,prerequisite_failed} in lorawan_app:start/2 line 19 in application_master:init/4 line 133
2017-07-19 04:48:54.526 [error] <0.120.0> CRASH REPORT Process <0.120.0> with 0 neighbours exited with reason: no match of right hand value {error,prerequisite_failed} in lorawan_app:start/2 line 19 in application_master:init/4 line 133
2017-07-19 04:53:24.165 [error] <0.120.0> CRASH REPORT Process <0.120.0> with 0 neighbours exited with reason: no match of right hand value {error,prerequisite_failed} in lorawan_app:start/2 line 19 in application_master:init/4 line 133

>

Rustie0125, 2017-07-19:
in the crash log 

> 2017-07-19 04:53:24 =CRASH REPORT====
  crasher:
    initial call: application_master:init/4
    pid: <0.120.0>
    registered_name: []
    exception exit: {{bad_return,{{lorawan_app,start,[normal,[]]},{'EXIT',{{badmatch,{error,prerequisite_failed}},[{lorawan_app,start,2,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_app.erl"},{line,19}]},{application_master,start_it_old,4,[{file,"application_master.erl"},{line,272}]}]}}}},[{application_master,init,4,[{file,"application_master.erl"},{line,133}]},{proc_lib,init_p_do_apply,3,[{file,"proc_lib.erl"},{line,237}]}]}
    ancestors: [<0.119.0>]
    messages: [{'EXIT',<0.121.0>,normal}]
    links: [<0.119.0>,<0.7.0>]
    dictionary: []
    trap_exit: true
    status: running
    heap_size: 987
    stack_size: 27
    reductions: 178
  neighbours:

>

Rustie0125, 2017-07-19:
result.

● lorawan-server.service - LoRaWAN Server
>
>    Loaded: loaded (/lib/systemd/system/lorawan-server.service; disabled)
>
>    Active: inactive (dead)
>
>
>

gotthardp, 2017-07-19:
Prerequisite failed means you have old Erlang. Version 19 or higher is required.

Rustie0125, 2017-07-19:
okay installing erlang again

Rustie0125, 2017-07-19:
So Erlang install reports " latest version installed "

when i run the server and check state this is result .

>
>> pi@raspberrypi:~ $ sudo systemctl status lorawan-server
>
> ● lorawan-server.service - LoRaWAN Server
>
>    Loaded: loaded (/lib/systemd/system/lorawan-server.service; disabled)
>
>    Active: active (running) since Wed 2017-07-19 05:47:06 UTC; 2s ago
>
>   Process: 6170 ExecStop=/usr/lib/lorawan-server/bin/lorawanctl stop
>> (code=exited, status=0/SUCCESS)
>
>  Main PID: 6245 (lorawan-server)
>
>    CGroup: /system.slice/lorawan-server.service
>
>            ├─6245 /bin/sh /usr/lib/lorawan-server/bin/lorawan-server
>
>            ├─6254 /usr/lib/erlang/erts-6.2/bin/beam.smp -Bd -- -root
>> /usr/lib/erlang -progname erl -- -home /var/lib/lorawan-server -- -noshell
>> -n...
>
>            └─6261 /usr/lib/erlang/erts-6.2/bin/epmd -daemon
>
>
>> Jul 19 05:47:06 raspberrypi systemd[1]: Started LoRaWAN Server.
>
> Jul 19 05:47:08 raspberrypi lorawan-server[6245]: {"init terminating in
>> do_boot",{{badmatch,{error,{lorawan_server,{bad_return,{{lorawan_ap...},{line
>
> Hint: Some lines were ellipsized, use -l to show in full.
>
>
>
When i check Status again, result is

>

> pi@raspberrypi:~ $ sudo systemctl status lorawan-server

● lorawan-server.service - LoRaWAN Server

   Loaded: loaded (/lib/systemd/system/lorawan-server.service; disabled)

   Active: failed (Result: exit-code) since Wed 2017-07-19 05:47:13 UTC;
> 1min 2s ago

  Process: 6290 ExecStop=/usr/lib/lorawan-server/bin/lorawanctl stop
> (code=exited, status=0/SUCCESS)

  Process: 6245 ExecStart=/usr/lib/lorawan-server/bin/lorawan-server
> (code=exited, status=1/FAILURE)

 Main PID: 6245 (code=exited, status=1/FAILURE)

> Jul 19 05:47:06 raspberrypi systemd[1]: Started LoRaWAN Server.

Jul 19 05:47:08 raspberrypi lorawan-server[6245]: {"init terminating in
> do_boot",{{badmatch,{error,{lorawan_server,{bad_return,{{lorawan_ap...},{line

Jul 19 05:47:12 raspberrypi lorawan-server[6245]: Crash dump was written
> to: erl_crash.dump

Jul 19 05:47:12 raspberrypi lorawan-server[6245]: init terminating in
> do_boot ()

Jul 19 05:47:12 raspberrypi systemd[1]: lorawan-server.service: main
> process exited, code=exited, status=1/FAILURE

Jul 19 05:47:12 raspberrypi lorawanctl[6290]: lorawan@raspberrypi command
> failed: nodedown

Jul 19 05:47:13 raspberrypi systemd[1]: Unit lorawan-server.service entered
> failed state.

Hint: Some lines were ellipsized, use -l to show in full.

gotthardp, 2017-07-19:
It means latest package not latest erlang. Here https://www.erlang-solutions.com/resources/download.html are instructions on how to install v19 or v20.

Rustie0125, 2017-07-19:
installed latest erlang but getting same error. going to start fresh
install, erlang first then lorawan server and see. will let you know

gotthardp, 2017-07-19:
To double check please type `erl`. This will type version info.

Rustie0125, 2017-07-19:
Hi Petr

I have no idea what is going on.
New install updated the pi, update erlang to 20 installed the lorawan.deb
when i run the command to start get the following.

>

Length: 2806738 (2.7M) [application/octet-stream]

Saving to: ‘lorawan-server_0.4.7_all.deb’

> lorawan-server_0.4.7_all.deb
>  100%[==========================================================================================================================================>]
>   2.68M   298KB/s   in 10s

> 2017-07-19 06:45:21 (265 KB/s) - ‘lorawan-server_0.4.7_all.deb’ saved
> [2806738/2806738]

> pi@raspberrypi:~ $ sudo erl

Erlang/OTP 20 [erts-9.0] [source] [smp:4:4] [ds:4:4:10] [async-threads:10]
> [hipe] [kernel-poll:false]

> Eshell V9.0  (abort with ^G)

1>

[2]+  Stopped                 sudo erl

pi@raspberrypi:~ $ sudo systemctl start lorawan-server

Failed to start lorawan-server.service: Unit lorawan-server.service failed
> to load: No such file or directory.

pi@raspberrypi:~ $ sudo systemctl start lorawan-server

Failed to start lorawan-server.service: Unit lorawan-server.service failed
> to load: No such file or directory.

pi@raspberrypi:~ $

Rustie0125, 2017-07-19:
After reboot

>> pi@raspberrypi:~ $ sudo systemctl start lorawan-server
>>
>>  pi@raspberrypi:~ $ sudo systemctl status lorawan-server
>
> ● lorawan-server.service - LoRaWAN Server
>
>    Loaded: loaded (/lib/systemd/system/lorawan-server.service; disabled)
>
>    Active: failed (Result: exit-code) since Wed 2017-07-19 07:12:37 UTC;
>> 29s ago
>
>   Process: 1199 ExecStop=/usr/lib/lorawan-server/bin/lorawanctl stop
>> (code=exite
>>    d, status=217/USER)
>
>   Process: 1197 ExecStart=/usr/lib/lorawan-server/bin/lorawan-server
>> (code=exite
>>    d, status=217/USER)
>
>  Main PID: 1197 (code=exited, status=217/USER)
>
>
>> Jul 19 07:12:37 raspberrypi systemd[1]: Started LoRaWAN Server.
>
> Jul 19 07:12:37 raspberrypi systemd[1]: lorawan-server.service: main
>> process...R
>
> Jul 19 07:12:37 raspberrypi systemd[1]: lorawan-server.service: control
>> proc...7
>
> Jul 19 07:12:37 raspberrypi systemd[1]: Unit lorawan-server.service
>> entered ....
>
> Hint: Some lines were ellipsized, use -l to show in full.
>
> pi@raspberrypi:~ $
>
>
>

gotthardp, 2017-07-19:
What the error log says, please?

Rustie0125, 2017-07-19:
Where are they located again ?

Rustie0125, 2017-07-19:
dont think the log files was created can not find them anywhere

gotthardp, 2017-07-19:
Logs should be in /var/lib/lorawan-server/...

Rustie0125, 2017-07-19:
there is no lorawan file under that path

Rustie0125, 2017-07-19:
used a new image on a new SD and tried the binary release installation
method. everthing installs fine when i run the server get the same error as
other method. as follows.

pi@raspberrypi:~/lorawan-server $ sudo bin/lorawan-server
> {"init terminating in
> do_boot",{{badmatch,{error,{lorawan_server,{bad_return,{{lorawan_app,start,[normal,[]]},{'EXIT',{{badmatch,{error,prerequisite_failed}},[{lorawan_app,start,2,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_app.erl"},{line,19}]},{application_master,start_it_old,4,[{file,"application_master.erl"},{line,272}]}]}}}}}}},[{lorawan_app,start,0,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_app.erl"},{line,16}]},{init,start_it,1,[]},{init,start_em,1,[]}]}}
> Crash dump was written to: erl_crash.dump
> init terminating in do_boot ()
> pi@raspberrypi:~/lorawan-server $

gotthardp, 2017-07-19:
The same error has the same reason-- old Erlang. Could it be you have two erlangs installed?

Rustie0125, 2017-07-19:
So I rolled back to 19.2 erlang on fresh install and it worked this time.

gotthardp, 2017-07-19:
Cool. Glad it works.

Rustie0125, 2017-07-20:
Hi Petr

Have you looked into writing a packet forwarder for the IC880A host system
? the ISMT exmaples does not have a packet forwarder and the TTN packet
forwarder seems over kill for what it needs todo, I have tried to modify
the TTN release to work with your server with no luck....

gotthardp, 2017-07-20:
The IC880A should be compatible with https://github.com/Lora-net/packet_forwarder. Also, the TTN gateway software is just a fork of that. All forks of the packet_forwarder are compatible with my server. I do not consider developing another GW software.

Rustie0125, 2017-07-20:
> packet_forwarder. Also, the TTN gateway software is just a fork of that.

i have tried setting up the global.conf file of the TTN fork to point at
the IP address of your server and 8080 port  but it does not register on
your server. The GW EUI has been updated to to the MAc generated EUI in the
Global file aswell where else does it need to be changed to work with your
private server ?

I have a IC880A with a Pi 2 Host succesfuuly connected to TTN, on the same
network your server is running, I have added the EUI to the server but not
sure what is required by "NetID" ?

gotthardp, 2017-07-20:
The port 1680 is used for the packet_forwarder

Rustie0125, 2017-07-20:
I Have tried 1680 before, and now again to confirm,
The Gateway EUI has been updated in the global and local json file. the
server is pointing at the local server IP address. when i reboot the system
it connects to TTN server but no luck on the local server.
Iv added a gateway on the local server with the same EUI namely"
  B827EBFFFEE37A71" but i have no idea what to input on NetID? and i see
its mandatory

gotthardp, 2017-07-20:
NetID can be e.g. `000000`.

Rustie0125, 2017-07-20:
No Response

have a look at the screen shots attached.
will the gateway be able to be connected to TTN and local server
simultaneously.?

thank you for the help

gotthardp, 2017-07-20:
The gateway can connect to one server only. (I don't see any screenshots attached,)

Rustie0125, 2017-07-20:
thats strange, they are attached ? PNGs

So What iv done sofar, the Server is running on a Pi plugged into the
network 192.168.11.17
the TTN packet forwarder is installed on another pi with the IC880A
gateway. in the /opt/ttn-gateway/packet_forwarder/basic_pkt_fwd directory
in the global_conf, I changed to the following

"gateway_conf": {
>         "gateway_ID": "B827EBFFFEE37A71",
>         /* change with default server address/ports, or overwrite in
> local_conf.json */
>         "server_address": "192.168.11.17",
>         "serv_port_up": 1680,
>         "serv_port_down": 1680,
>         /* adjust the following parameters for your network */
>         "keepalive_interval": 10,
>         "stat_interval": 30,
>         "push_timeout_ms": 100,
>         /* forward only valid packets */
>         "forward_crc_valid": true,
>         "forward_crc_error": false,
>         "forward_crc_disabled": false
>     }
> }

Saved and restarted the the gateway with connection to the internet turned
off (NO TTN access)  but still no luck on the server.

Rustie0125, 2017-07-21:
Hi Petr

I managed to get the gateway talking to the private server, I have
registered a node on the server and can see in the log that there is errors
in the config like "bad_mic" is there a manual for the web application or
tutorial on the interface?

gotthardp, 2017-07-21:
There is nothing beyond the documents in the `doc` folder. The error you have means you misconfigured the security keys.

Rustie0125, 2017-07-21:
Okay thank you.

I am in the process of trying the binary for windows. I have installed
Erlang. but the instruction does not explain how to install or unpack the
lorawan server on windows. running bin/lorawan-service.bat in erlang does
nothing as expected

gotthardp, 2017-07-21:
There are instructions even for Windows: https://github.com/gotthardp/lorawan-server/blob/master/doc/Installation.md#using-the-binary-release-on-windows

Rustie0125, 2017-07-21:
> There are instructions even for Windows: https://github.com/gotthardp/
> lorawan-server/blob/master/doc/Installation.md#using-the-
> binary-release-on-windows

Yes that is the instructions that make no sense and is incomplete

gotthardp, 2017-07-21:
Nothing is perfect, but comments like yours are not constructive. Several people successfully installed the server based on these instructions, so they are not rubbish. If you want me to help you, you should help me by providing following the available guidelines and let me know in which step you got some failure.

Rustie0125, 2017-07-21:
Petr 

I understand that but sending you mail and saying that the instructions is not sufficient and you sending the same link back saying here is the instruction is not helpful either. 

You binary windows instructions says 
1) "install Erlang"
2) "unpack the release " what release ? Link? 
After unzipping what do I do with that file? Double click on bat file? Past folder into specific directory ?
3) " run by " bin/lorawan-server.bat
Copy and past " bin/LoraWAN-server.bat" into Erlang?
Copy and past into win CMD? 
After bat file is run. 
How do you access the webpage ?what will default IP be?

Server configuration. 
Does the LoraWAN-server/releases/<VERSION>/says.config refer to windows build , Linux or Mac ? 

The over all steps is incomplete to help a noob getting the server running. 

Please safe our environment and do not print this email.

gotthardp, 2017-07-21:
I made some amendments to the documentation to make it more clear:
 * the binaries can be unpacked to any folder (e.g. lorawan-server), but currently the path must not contain spaces (issue #158).
 * start by doubke-clicking on `lorawan-server.bat` in `lorawan-server/bin`
 * access the webpage at `http://127.0.0.1:8080`.
 * server configuration refers to all builds

closed 2017-07-29 by gotthardp

Rustie0125, 2017-08-09:
This Issues with the documentation is still not solved. I Have tried running the windows based build and after a week of not being able to locate the file as per your directions in the "lorawan-server/bin" location I ran the .bat in the script folder which resulted in windows being corrupted and crashed my complete PC. I have now finally gotten back online and the same question still remains. please update the documentation on how to find, install and use the Binary file as none of the available files for download results in a lorawan-server/Bin directory.

Rustie0125, 2017-08-09:
Here is the screenshot of my release unzipped on the desktop. no BIN folder is present.
[image]

gotthardp, 2017-08-09:
Hello. You need to unpack the binary release, i.e. the `.tar.gz` file. I just clarified this in the documentation.

Rustie0125, 2017-08-10:
Thanx Petr 

Now it works.thank you for the clarification. 

I am working with two different Nodes(devices) I have added the nodes to the nodes list (see screen shot) but I keep getting a "BAD_MIC"  or a "	{unexpected_data,{rxdata,15,10,<<74,23>>,false,true}}" error 

[image]
[image]
[image]

The one node is a standard microchip mote RN2483 and the other is a Adafruit Feather.

gotthardp, 2017-08-10:
bad_mic means the NwkSKey or AppSKey is not matching the keys configured in the device. Double check you are using the right keys.

unexpected_data means the keys are configured well, but you set the application to 'SemtechMote' or 'MicrochipMote' although your device is sending some other data. Change the 'Application' field to 'WebSocket' for example to get rid of this.

Rustie0125, 2017-08-10:
Thank you 

Changed to websocket but now it reports "{uplinks_lost,11}"

gotthardp, 2017-08-10:
That doesn't have to be an error, it could also mean that the gateway or server was off while the device was sending data. It's rather "uplinks_missed".

Rustie0125, 2017-08-10:
Okay, I see. Im trying to send the received packet from the server via a MQTT message to a MQTT server. do you have an example of someone that has done this ? I did look at this page https://github.com/gotthardp/lorawan-server/blob/master/doc/Integration.md but can not seem to get it to work on windows, the Mosqiuto project latest download has errors when trying to install and Rabit is to complicated to even try.

gotthardp, 2017-08-10:
RabbitMQ works well with Windows. If you want MQTT broker on windows, use that.

altishchenko, 2017-08-10:
Or you can use emq from http://emqtt.io
LoRaWAN server uses its client (emqttc) to build mqtt connections in the background.

The one thing to look out - is the contents of the data you send. Some MQTT servers which are combined with sort of visualizing app and most of the free portals out there, like Adafruit IO, will insist on the JSON structure of their own which is not yet implemented currently.

Cheers,
Alexander Tishchenko
mob: +7 (916) 704-61-57
atishch@gmail.com

[34 more replies]