gotthardp/lorawan-server #162: How to authenticate for the REST api.

gjoosen, 2017-07-14:
Hello,

How should I authenticate for the REST api.
I'm using postman to test the REST api, but when I try the following url: `IP:8080/users` I get the following status back: 401 Unauthorized. Is there some kind of JWT to authenticate, I cannot find anything about this in the documentation.

gotthardp, 2017-07-14:
It's pure HTTP Basic authentication. the tool shall reply to 401 with the password. did it?

gjoosen, 2017-07-14:
The tool doesn't ask for a password or something, just returns 401.
Here is a screenshot of the tool return.
[image]

gotthardp, 2017-07-15:
You have to enter the password before sending the request:
https://www.getpostman.com/docs/postman/sending_api_requests/authorization

gjoosen, 2017-07-15:
Thanks, solved the problem!

closed 2017-07-15 by gjoosen