gotthardp/lorawan-server #163: How to change the database directory

stephanedavid, 2017-07-18:
I have made an installation using the debian package and i would like to know if it is possible to change the installation directory for the mnesia database and the log files.
I'm using a read only root file system and i need to put the mnesia and logs to a rw partition, i have create a environment variable  LORAWAN_HOME=/home/ogate/lorawan-server in /etc/environment without any succes.

gotthardp, 2017-07-18:
Have a look at /lib/systemd/system/lorawan-server.service and set the LORAWAN_HOME there (https://github.com/gotthardp/lorawan-server/blob/master/scripts/lorawan-server.service#L7).

stephanedavid, 2017-07-19:
Thank you Petr for your help.
Now your server works very well on our lorawan gateway.

closed 2017-07-19 by stephanedavid