gotthardp/lorawan-server #164: Question on FCntDown counter incrementing during the first uplink message sent after join success

frozzygoh, 2017-07-18:
Hi Gotthard,

I have a question regarding FCntDown incrementation.

In my LNS, I have: 
- disabled all connectors to external applications, 
- Under device being registered, 'FCnt Check' is set to "disabled".

**Scenario**
After join request and accept step, upon sending a message on the uplink from device towards the LNS, it is observed that the FCntUp counter remains as 0, and FCntDown counter is incremented by 1.  Subsequent uplink messages sent from device to LNS only affects the FCntUp counter to increment which is expected.

I have the observations steps are like this:
1) After join request and accept.
FCntUp: 0 , FCntDown: 0
2) 1st uplink message sent:
FCntUp: 0, FCntDown: **1**
3) 2nd uplink message sent:
FCntUp: 1, FCntDown: 1
4) 3rd uplink message sent:
FCntUp: 2, FCntDown: 1

Do you have any insights on the observation in step 2), on why FCntDown was incremented, when in fact, there was no downlink message sent? Is there any in-build applications with the lorawan-server which by default is set to send a downlink message?

Thank you in advance!

gotthardp, 2017-07-18:
Hello. I guess the MAC layer decided to send a downlink, e.g. requesting the device status (the reason can be seen in the log). And that's why the FCntDown was incremented.

frozzygoh, 2017-07-19:
Thank you Peter for the insights! :) Yes indeed, I notice in the logs on mac command corresponding to a dev_status_req.

gotthardp, 2017-07-19:
It can be disabled on the Status tab if you dont like it.

frozzygoh, 2017-07-21:
Oh I see, I will have a try and see. 
Thank you Petr!

frozzygoh, 2017-07-28:
Hello Petr,

May I check with you where and what should be the inputs to the Status tab to: 
- Disable the device status request upon network join/reset, and 
- Disable the default time/fcnt intervals defined in the devstat_gap?
- Disable device status request when ADR parameters change?

I see that the Status tab fields have been populated with the received device status, but not sure which field I should change to disable the device status request.

Thank you!

gotthardp, 2017-07-29:
The 'Request Status' should be set to 'False'. This disables (should disable) all device status requests, including those after Join, those in time/fcnt intervals as well as those after ADR. Didn't it disable it?

closed 2017-08-10 by gotthardp

frozzygoh, 2017-08-11:
Hello Petr, 

Sorry I realized my reply didn't get posted. I still wasn't able to find the 'Request Status' field under Devices -> Status or Nodes :P So, I hadn't had the chance to try it yet... 

Thanks a lot for all your help so far!

gotthardp, 2017-08-11:
Did you install the latest release 0.4.8? You should see it there.

frozzygoh, 2017-08-11:
Ok I will check that latest version. Thank you Petr!