gotthardp/lorawan-server #166: fcnt gap too large error

kammyy, 2017-07-26:
Hi,

I am getting this error for a device.

The current scenario:
"My lorawan gateway was switched OFF for a 3 days while few end nodes were sending data for every 2 minutes" . When the gateway was switched ON , then i am receiving "{fcnt_gap_too_large,51}" for all my devices. This number "51" keeps on increasing for every uplink from device.

What to do in this scenario ?

Cant the server accept the messages only by showing "{uplinks_lost,--}" ?

REgards

gotthardp, 2017-07-26:
Hello. You can temporarily disable the FCntCheck. With the first message the FCnt check will be updated and you will be able to switch the FCnt check to its original setting. Would that work for you?

kammyy, 2017-07-26:
Yeah it worked. Thanks.

One more query:
"For some device i see the error {uplinks_lost,--} (after this the server continues to accept data from the device) but for some devices i see the error {fcnt_gap_too_large,---} (and the server does not accept messages from the device after that) ."

Please explain the two outcomes.

gotthardp, 2017-07-26:
well, uplinks lost means some uplink frames were not received (from some reason) by the server.
fcntgaptoolarge means too many frames were lost, or device reset the counter

kammyy, 2017-07-26:
Is there any quantification on the number of uplinks lost after which the server puts it in category of fcntgaptoolarge ?

If i have selected "Reset on Zero" then should not the server start the fcnt again if the device reset the counter ?

Regards

gotthardp, 2017-07-26:
Max gap is 16384 (defined in the lorawan standard), so a lot.

With reset on zero the server will start again, but the packets after reset must not be lost. The "max_lost_after_reset" is 10, so significantly less but this value can be set in the config.

kammyy, 2017-07-26:
Where is that config ?

Regards

gotthardp, 2017-07-27:
Have a look in https://github.com/gotthardp/lorawan-server/blob/master/src/lorawan_server.app.src and then add a `max_lost_after_reset` line to the conf.sys.

closed 2017-08-10 by gotthardp