gotthardp/lorawan-server #168: server shutdown

Ankitjd18, 2017-07-30:
Hi,
Server shut down with below log messages. Can you tell us what happened to the server? Which case failed here.
crash.log - 
`2017-07-29 15:26:56 =SUPERVISOR REPORT====
     Supervisor: {local,lorawan_sup}
     Context:    child_terminated
     Reason:     {{case_clause,<<"FSK">>},[{lorawan_gw_forwarder,parse_rxpk,1,[{file,"/home/ubuntu/lorawan-server/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_gw_forwarder.erl"},{line,158}]},{lorawan_gw_forwarder,'-rxpk/2-fun-0-',3,[{file,"/home/ubuntu/lorawan-server/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_gw_forwarder.erl"},{line,152}]},{lists,map,2,[{file,"lists.erl"},{line,1239}]},{lorawan_gw_forwarder,rxpk,2,[{file,"/home/ubuntu/lorawan-server/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_gw_forwarder.erl"},{line,150}]},{lists,foreach,2,[{file,"lists.erl"},{line,1338}]},{lorawan_gw_forwarder,handle_info,2,[{file,"/home/ubuntu/lorawan-server/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_gw_forwarder.erl"},{line,54}]},{gen_server,try_dispatch,4,[{file,"gen_server.erl"},{line,616}]},{gen_server,handle_msg,6,[{file,"gen_server.erl"},{line,686}]}]}
     Offender:   [{pid,<0.23027.0>},{id,packet_forwarder},{mfargs,{lorawan_gw_forwarder,start_link,[[{port,1680}]]}},{restart_type,permanent},{shutdown,5000},{child_type,worker}]

2017-07-29 15:26:56 =SUPERVISOR REPORT====
     Supervisor: {local,lorawan_sup}
     Context:    shutdown
     Reason:     reached_max_restart_intensity
     Offender:   [{pid,<0.23027.0>},{id,packet_forwarder},{mfargs,{lorawan_gw_forwarder,start_link,[[{port,1680}]]}},{restart_type,permanent},{shutdown,5000},{child_type,worker}]`

error.log - 
`2017-07-29 15:26:56.914 [error] <0.276.0> Supervisor lorawan_sup had child packet_forwarder started with lorawan_gw_forwarder:start_link([{port,1680}]) at <0.23027.0> exit with reason no case clause matching <<"FSK">> in lorawan_gw_forwarder:parse_rxpk/1 line 158 in context child_terminated
2017-07-29 15:26:56.914 [error] <0.276.0> Supervisor lorawan_sup had child packet_forwarder started with lorawan_gw_forwarder:start_link([{port,1680}]) at <0.23027.0> exit with reason reached_max_restart_intensity in context shutdown`

debug.log -
`2017-07-29 15:26:56.914 [error] <0.276.0> Supervisor lorawan_sup had child packet_forwarder started with lorawan_gw_forwarder:start_link([{port,1680}]) at <0.23027.0> exit with reason no case clause matching <<"FSK">> in lorawan_gw_forwarder:parse_rxpk/1 line 158 in context child_terminated
2017-07-29 15:26:56.914 [debug] <0.306.0>@lorawan_connector_mqtt:handle_info:110 Unknown message: {'EXIT',<0.305.0>,shutdown}
2017-07-29 15:26:56.914 [error] <0.276.0> Supervisor lorawan_sup had child packet_forwarder started with lorawan_gw_forwarder:start_link([{port,1680}]) at <0.23027.0> exit with reason reached_max_restart_intensity in context shutdown
2017-07-29 15:26:56.914 [info] <0.278.0>@lorawan_gw_router:terminate:120 gateway router terminated: shutdown
2017-07-29 15:26:56.915 [info] <0.33.0> Application lorawan_server exited with reason: shutdown`

gotthardp, 2017-07-30:
Oh, what device sent this frame and in what situation? The server (currently) does not support FSK frames (and your device sent one). Of course, the crash is a bug, but I wonder if I should implement FSK handling or if a protection against FSK is enough.

gotthardp, 2017-08-10:
The crash is fixed in 0.4.8

closed 2017-08-10 by gotthardp