gotthardp/lorawan-server #170: Websockets

Kralg, 2017-07-31:
Hello,

I am trying to run the ws.html example. Can you explain on how to do that? I downloaded the admin folder, edit the ws.html file and change `wsHost = "ws://" + 127.0.0.1:8080 + "/ws/" + $("#type").val() + "/" + $("#server").val() + "/" + $("#format").val();` I close the file and I double click it. However when I click connect Connecting window doesn't show. What is the right way to do that? Some help would be very apreciated.

gotthardp, 2017-07-31:
You can connect to ws.html directly by `http://server:8080/admin/ws.html`. You know that, right? The page should display fields where you need to enter DevAddr and desired content type.

Kralg, 2017-07-31:
I think that once I found it. I was mislead this time due to the reason that main dashboard is under admin#/ (e.g. :8080/admin#/nodes/list) and I was trying to do the same admin#/ws.html and some different ways. Maybe it would be good to write in documentation server:8080/admin/ws.html not only admin/ws.html.

Another thing, I have created the handler with Build Downlink fun(_Port, #{cnt := CNT}) ->
  <<CNT>>
end.
from websocket I am sending {"cnt":4, "port":2}, in nodes panel I can see that in downlinks there is new line with port address 2 but there is no txdata. Can you tell me what is wrong here?

By the way, thanks for 0.4.8!

gotthardp, 2017-07-31:
You should send ` {"fields":{"cnt":4}, "port":2}`.

Kralg, 2017-07-31:
Somehow still there is no txdata

gotthardp, 2017-07-31:
What does the debug.log say?

Kralg, 2017-07-31:
Actually, it works. Meanwhile I was trying different json's and the point was in mismatch. 
Thanks.

gotthardp, 2017-07-31:
Cool :)

closed 2017-07-31 by gotthardp