gotthardp/lorawan-server #171: HTTP backend

Kralg, 2017-08-01:
Hello,

I am trying to use HTTP and TXframes REST command. Am I supposed to send all these field as listed in Downlinks part? https://github.com/gotthardp/lorawan-server/blob/master/doc/JSON.md

gotthardp, 2017-08-01:
Hello. Thanks for testing the server.
1) The HTTP backed works for uplinks only for now. Connector for downlinks is yet to be done.
2) The SNR shown in Nodes list is the **downlink** SNR reported by the device, whereas the SNR in received frames is the **uplink** SNR.
3) The state change is a bug, See issue #172.
4) You are right. The newer versions alternate RX1/RX2 slots during JOIN attempts.
5) I can display the version number. See issue #173.
6) The display window may not show because you are running an old version, or because the page is cached. Sometimes it needs few hours to see the updated page.
7) Yes, some figures are outdated and should be refreshed.

Kralg, 2017-08-02:
Hello, 

Thank you for the answers. 
1. I thought it was meant to be implemented in 0.4.8. Maybe I misunderstood the HTTP backend chat. Are you planning to implement that? If so when it would be possible to have it?
2. I see. Maybe it would be great to have some popup windows with information once hovered over SNR or so. Probably other columns are self explaining. 
4. I am wondering why server first uses Rx2 in auto mode instead of Rx1 if device sends data on Rx1? Is this has some logic behind?

gotthardp, 2017-08-02:
1) At least a month. I am currently overloaded.
2) I just commited a fix.
4) Are you sure? My tests show that the first JOIN Accept is done in RX1.
6) Please do.

Kralg, 2017-08-02:
Damn, it is so long :)
I just have checked which server I am using by turning it off and checking if on other computers it is down and restarting it again. I also tried to create a device on RPi and it occurs on other computers right away. 
I really start server from tar -zxvf 0.4.8 version bin folder with sudo ./lorawan-server. Also on other computers there is no blue icon in internet tab which I can see in RPI.

gotthardp, 2017-08-02:
3) Could you have a look in the log file to see if it reports RX1 or RX2 during join, please?
6) I really suspect there is an old version or a cache somewhere. The blue icon is another thing I added recently.
7) This is exactly how it works :-) If there are more frames to send the server sets the pending flag even if it was not set in the payload.

Kralg, 2017-08-02:
3 I will go back to it a bit later.
6 And so, you right. F5 is not enough ctrl+F5 really did the trick and webpage is fully updated. Sorry for this, haven't had this before.

Kralg, 2017-08-03:
3 I just looked to log, it seems first to be RX1:
`2017-08-03 13:13:56.486 [info] <0.305.0>@lorawan_mac:handle_join:170 JOIN REQUEST 70B3D57EF000425C 000000000000000A -> 45A48C6A
2017-08-03 13:13:56.487 [debug] <0.305.0>@lorawan_mac:handle_join:190 Join-Accept in RX1: {0,0,869.525}
2017-08-03 13:14:11.335 [warning] <0.305.0> node 45A48C6A {repeated_reset,1}
2017-08-03 13:14:11.336 [info] <0.305.0>@lorawan_mac:handle_join:170 JOIN REQUEST 70B3D57EF000425C 000000000000000A -> 45A48C6A
2017-08-03 13:14:11.336 [debug] <0.305.0>@lorawan_mac:handle_join:193 Join-Accept in RX2: {0,0,869.525}
2017-08-03 13:14:26.194 [warning] <0.305.0> node 45A48C6A {repeated_reset,2}
2017-08-03 13:14:26.195 [info] <0.305.0>@lorawan_mac:handle_join:170 JOIN REQUEST 70B3D57EF000425C 000000000000000A -> 45A48C6A
2017-08-03 13:14:26.195 [debug] <0.305.0>@lorawan_mac:handle_join:190 Join-Accept in RX1: {0,0,869.525}`
8 It would be also very good to be able to delete data from mnesia of old devices which are no more active. Or to delete data of each device or so.

Kralg, 2017-08-09:
Does server support downlinks transmition one at the time? I send multiple {"data":"1253", "port":255, "pending":false} from websocket but server sends them all immediately. Shouldn't it send frames only one per end-device transmition if "pending":false ? Also if I write "pending":**"**false**"** frames disappear from the list one by one, but they are not transmitted.

gotthardp, 2017-08-09:
Please keep this thread related to "HTTP backend for downlinks" and raise other questions as separate issues. It's becoming hard to follow all the sub-threads.

Isn't 10 duplicate of 7?

Kralg, 2017-08-10:
Ok, I will. Well not really in 10 I am not using handlers (I think so because there is no "fields" parameter) and therefore I would expect server to read pending flag and if it is false then send only one downlink per one end-device transmition (do not inform end-device that there are more packets to transmit right away) and if it is true then all downlinks send right away. I thought that due to the handler implementation flag get's set automatically, but if I send only "data" parameter I should be able to manually define pending flag. Otherwise is there any meaning to have pending flag if it is reset automatically by server and I am not able to define it myself?

gotthardp, 2017-08-10:
10  - the pending handling was confusing, I agree. I just changed that, so with the latest 'master' you can send pending=false even if there are some data in the queue.

gotthardp, 2017-08-10:
Just for the record-- the only issue that remains unresolved is the HTTP interface for downlinks (sub-issue 1).

Kralg, 2017-08-14:
Alright but what about RX2 window?

gotthardp, 2017-08-14:
The log you provided, which showed RX2 was after a repeated join. In auto-mode the very first Join-Accept after the device was sending some frames is sent in RX1. The subsequent are alternated between RX2 and RX1.
If your device didn't sent any frame and just joined, the second join is considered as a re-join and RX2 is used. This logic can be disabled by pernamently selecting RX1 or RX2.

But please keep this issue focused on HTTP dowlink. If you believe there is an issue in the RX2 window handling, please open another issue related to that.

Kralg, 2017-08-14:
I see. When would it be possible to expect the 0.4.9 version of the server? Still 2 weeks?

gotthardp, 2017-08-14:
This week. I want to implement the downlinks before releasing.

gotthardp, 2017-08-18:
Hi there. I just did the release, but without http downlink. It needs more thoughts, so I need more time to think about it.

Kralg, 2017-09-04:
Hi, are there any news on this topic?

gotthardp, 2017-09-04:
Nope. The required refactoring is not easy and I am busy with tasks of higher priority.

gotthardp, 2018-01-05:
Sorry for the delay. I just pushed the implementation to the devel branch. It will be included in the famous 0.5.0.

closed 2018-01-05 by gotthardp