gotthardp/lorawan-server #174: Ability to use funs with RAW

altishchenko, 2017-08-07:
Hello Petr,
I think it would be a great idea to be able to use Up/Down funs with RAW mode, might help a lot things out there. The call-reply format should be binary-in (possibly accompanied by a map of gateway/devaddr/deveui etc) --> binary-out. Same applies to downlinks also.

gotthardp, 2017-08-07:
Oh, yes. Binary-in, binary-out is not supported, but can be done quite easily.

altishchenko, 2017-08-07:
Exactly what I ask :) ! This way I can even fool Adafruit :)

gotthardp, 2017-08-11:
I did a basic implementation, preserving the API (so without any maps, just port and data). Will you test it, please?

altishchenko, 2017-08-13:
Sorry, fell out for a while. Will test it straight away!

altishchenko, 2017-08-13:
It just works great! Thank you Petr!!!

closed 2017-08-13 by altishchenko