gotthardp/lorawan-server #175: Communication with Azure

gradzik, 2017-08-08:
Hi,

I have a problem with lorawan-server and ms azure. I added gateway, device and connector in server dashboard. I can see frames in "Received Frames", mqtt connection with azure works fine but with every message from a device I get these warnings:

2017-08-08 08:22:51.482 [warning] <0.246.0> node 12345678 {uplinks_lost,13}
2017-08-08 08:22:51.487 [debug] <0.246.0>@lorawan_mac_commands:build_fopts:42 12345678 <- [dev_status_req]
2017-08-08 08:22:51.741 [warning] <0.240.0>@lorawan_gw_router:handle_cast:88 Downlink request ignored. Gateway <<184,39,235,255,255,67,161,189>> not connected.
2017-08-08 08:22:51.864 [warning] <0.240.0> gateway B827EBFFFF43A1BD {uplinks_lost,1}

And I can't send anything from azure to device. When sending a message from device explorer (azure) I get:

Message Feedback status: "Success", Description: "Success"

But lorawan-server shows these errors:

2017-08-08 06:43:18.980 [error] <0.262.0>@lorawan_connector_mqtt:match_vars:246 Topic <<.....>> does not match pattern {<<.....>>,[]}
2017-08-08 06:43:18.980 [error] <0.262.0>@lorawan_application_backend:send_downlink:147 Unknown downlink target: #{}

altishchenko, 2017-08-08:
What do you have in a 'Received Topic' pattern? (Or Consumed as it was called before)

gradzik, 2017-08-08:
Received from azure:
<<devices/BE7A0000000017D8/messages/devicebound/%24.mid=e58c9eea-6e4d-430c-844a-5247670e7d8d&%24.to=%2Fdevices%2FBE7A0000000017D8%2Fmessages%2FdeviceBound&iothub-ack=full&2=3>>

Pattern:
<<devices/BE7A0000000017D8/messages/devicebound/#>>

altishchenko, 2017-08-08:
Is BE7A0000000017D8 your deveui? If it is, you Received pattern should look like 'devices/{deveui}/...'
This way the server will know which one of the devices using this backend shall get a reply.

gradzik, 2017-08-08:
BE7A0000000017D8 is Client ID and Device ID

altishchenko, 2017-08-08:
Well, then give a try to my earlier proposal - devices/{deveui}/.... in received topic

gotthardp, 2017-08-08:
The server did not support partial matching and wildcards in patterns. I just fixed that. Wait for next release, or build the server from the 'master' and then try `devices/{deveui}/messages/devicebound/*` (note the star at the end).

gotthardp, 2017-08-08:
Update: # should be used as a "wildcard" to be consistent.

gradzik, 2017-08-11:
Thank you for help I think I know how it works :)

closed 2017-08-11 by gotthardp