gotthardp/lorawan-server #177: Issue joining node over OTAA

Sviests, 2017-08-08:
I am having issues registering a new device to my network using OTAA. I have had this device registered before and there were no issues. I have deleted the device and I have added it back with the correct keys, but I am getting this error that I`m unable to parse.

`2017-08-08 16:07:44.325 [info] <0.288.0>@lorawan_mac:handle_join:170 JOIN REQUEST 70B3D5B020000039 70B3D5B020035F16 -> 56DBA356
2017-08-08 16:07:44.326 [error] <0.288.0>@lorawan_worker:handle_cast:31 {function_clause,[{lorawan_mac,initial_rxwin,[undefined,{0,0,869.525}],[{file,"/home/sviests/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac.erl"},{line,209}]},{lorawan_mac,handle_join,6,[{file,"/home/sviests/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac.erl"},{line,180}]},{lorawan_worker,handle_cast,2,[{file,"/home/sviests/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_worker.erl"},{line,22}]},{wpool_utils,do_try,1,[{file,"/home/sviests/lorawan-server/_build/default/lib/worker_pool/src/wpool_utils.erl"},{line,59}]},{wpool_process,handle_cast,2,[{file,"/home/sviests/lorawan-server/_build/default/lib/worker_pool/src/wpool_process.erl"},{line,155}]},{gen_server,try_dispatch,4,[{file,"gen_server.erl"},{line,616}]},{gen_server,handle_msg,6,[{file,"gen_server.erl"},{line,686}]},{proc_lib,init_p_do_apply,3,[{file,"proc_lib.erl"},{line,247}]}]}`

Sviests, 2017-08-08:
The issue was that I had not provided the txwin field in the JSON when creating the device using the API. Is this a required field?

gotthardp, 2017-08-08:
You used rest API directly to create the device, right? The the txwin field shouldn't be mandatory; I just made a fix.

Sviests, 2017-08-08:
Yes, I used the rest API. Thanks for the fix!

closed 2017-08-08 by gotthardp