gotthardp/lorawan-server #178: Error 400 on POST request

Jorrie911, 2017-08-10:
Hey,

Great work on the Lora server. I just have a question regarding the POST requests.
When I try to POST data to a server (http://posttestserver.com/post.php for testing purposes). I get a Error 400.
Below is some extract from ngrep

T 10.0.0.173:54937 -> 107.170.47.130:80 [AP]
POST  HTTP/1.1..content-length: 146..host: posttestserver.com:80..content-type: application/x-www-form-urlencoded....appargs=Arg&data=%00%00%00%00%00%17%00%17%00C%0C%CC%0C%CC%00%01%00%00%FE%00%FE%00%FE%00%FE%00%FF%1E%1E%00%00%00%00%00%
00%00%00%04%00Y%8Cf%FD%00%00
##
T 107.170.47.130:80 -> 10.0.0.173:54937 [AP]
HTTP/1.1 400 Bad Request..Date: Thu, 10 Aug 2017 12:00:40 GMT..Server: Apache/2.4.18 (Ubuntu)..Content-Length: 311..Connection: close..Content-Type: text/html; charset=iso-8859-1....<!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">.<html><head>.<title>400 Bad Request</title>.</head><body>.<h1>Bad Request</h1>.<p>Your browser sent a request that this server could not understand.<br />.</p>.<hr>.<address>Apache/2.4.18 (Ubuntu) Server at doptsv1.localdomain Port 80</address>.</body></html>.
#####
Any help will be appreciated, I am quite new to the Lora communication and also Linux, so I might be doing something wrong

Thanks

gotthardp, 2017-08-10:
Hi. To what URI are you posting and what you ware trying to achieve?

Jorrie911, 2017-08-10:
I am posting to http://posttestserver.com/post.php(just for testing now) then I can see the data on
http://posttestserver.com/data/

When this works successfully I will create a php file on my own server to process the data and put the data into a SQL database

gotthardp, 2017-08-10:
I still don't see how this is related to lorawan-server. Could you please explain this a bit more?

Jorrie911, 2017-08-10:
When I post with the lorawan-server I get a Error 400, which indicates that there might be a problem with the HTTP header or data that has to be send.
This is found in the debug.log:
2017-08-10 15:38:14.797 [debug] <0.360.0>@lorawan_connector_http:handle_info:83 HTTP request failed: 400

Using some other way of posting(such as hurl.it) to test the posttestserver.com the connection is successful (200) and the data is received on the other end

gotthardp, 2017-08-10:
But again: what URI on the lorawan-server are you POSTing to?

Jorrie911, 2017-08-10:
[image]

This is what my setup looks like

gotthardp, 2017-08-10:
I see now. It may be because of the data attribute. You defined some 'Parse Uplink' handler function, right?

Jorrie911, 2017-08-11:
Yes,
Since I am not that fimiliar with fun, I used a sample from another issue I found:
fun(_Port, Bin) ->
#{data => binary_to_list(Bin)}
end.

It seems to work well because as you can see in the ngrep extract above:
data=%00%00%00%00%00%17%00%17%00C%0C%CC%0C%CC%00%01%00%00%FE%00%FE%00%FE%00%FE%00%FF%1E%1E%00%00%00%00%00%
00%00%00%04%00Y%8Cf%FD%00%00

This is correct this is my data

gotthardp, 2017-08-11:
You may be getting the error because the data are binary. Have you tried Raw payload instead?

Jorrie911, 2017-08-11:
Could you please give an example of how the erlang code will look for a Raw payload?

gotthardp, 2017-08-11:
Just switch the handler to Raw. No functions are used there.

Jorrie911, 2017-08-11:
With the RAW the data doesn't display correctly as seen below
This is the ngrep
T 10.0.0.173:46771 -> 107.170.47.130:80 [AP]
  POST  HTTP/1.1..content-length: 45..host: posttestserver.com:80..content-type: application/octet-stream.............C.............................Y.{~..
##
T 107.170.47.130:80 -> 10.0.0.173:46771 [AP]
  HTTP/1.1 400 Bad Request..Date: Fri, 11 Aug 2017 07:41:00 GMT..Server: Apache/2.4.18 (Ubuntu)..Content-Length: 311..Connection: close..Content-Type: text/
  html; charset=iso-8859-1....<!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">.<html><head>.<title>400 Bad Request</title>.</head><body>.<h1>Bad Request</
  h1>.<p>Your browser sent a request that this server could not understand.<br />.</p>.<hr>.<address>Apache/2.4.18 (Ubuntu) Server at doptsv1.localdomain Po
  rt 80</address>.</body></html>.
########

gotthardp, 2017-08-11:
Well, but this is an issue of the other server. It should be able to accept application/octet-stream.

Jorrie911, 2017-08-14:
Using hurl.it to simulate a POST the server accepts application/octet-stream
[image]

gotthardp, 2017-08-14:
Good to hear. Thanks for letting me know.

closed 2017-08-14 by gotthardp