gotthardp/lorawan-server #180: Notes editbox

Kralg, 2017-08-14:
Hi, would it be possible to add some kind of notes in devices, node and gateway edit window?
I have some additional information about them and what I am doing right now I am saving that information in local db. I suppose it could be possible to add some kind of erlang application which would use mnesia db and store additional data in it, but maybe notes editbox would be simpler and handy option for the server. My additional data right now can have around 300 symbols or so.

gotthardp, 2017-08-14:
Hello. The 'Group' and 'Arguments' fields are intended for some application related data. These are not sufficient for you?

Kralg, 2017-08-14:
> Arguments, which is an opaque string with application-specific settings. 

So server is not using Arguments anywhere just saves it and that is it? This probably would work.
Group I am using to group the devices and send multiple downlinks to all of them.

gotthardp, 2017-08-14:
Correct. Server just stores the values and sends the values to backend applications.

closed 2017-08-16 by Kralg