gotthardp/lorawan-server #181: Rx1 and Rx2 windows

Kralg, 2017-08-14:
> The log you provided, which showed RX2 was after a repeated join. In auto-mode the very first Join-Accept after the device was sending some frames is sent in RX1. The subsequent are alternated between RX2 and RX1.
> If your device didn't sent any frame and just joined, the second join is considered as a re-join and RX2 is used. This logic can be disabled by pernamently selecting RX1 or RX2.
> 

What we were doing was to turn off and turn on the power of the end device after the registration was successful. The logic of the server you wrote then really would do the same as we see, but I think it would be good if server would reset the counter(Rx1, Rx2) after successful connection. This would also make sense, that Rx1 was successful and there is no need next time to use Rx2 for the same device.

gotthardp, 2017-08-14:
The problem is that the server doesn't know the Join was successfully received by the node until the node first sends a frame.

Kralg, 2017-08-16:
I see. Would it be hard for server to trigger weather or not end-device sent first packet? And reset the switching?

gotthardp, 2017-08-16:
Yes, join after a successfully received frame is not considered as re-join, so its done in RX1.

closed 2017-08-23 by gotthardp