gotthardp/lorawan-server #184: server on raspberry pi 2

oyj0594, 2017-08-16:
Hi, i installed erlang 19.2 and lora server.
but when i executed, log came up like below..
what's the problem?? 

best regards.

> ./lorawan-server
> {"init terminating in do_boot",{{badmatch,{error,{lorawan_server,{bad_return,{{lorawan_app,start,[normal,[]]},{'EXIT',{{badmatch,{error,eaddrinuse}},[{lorawan_app,start,2,                                   [{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_app.erl"},{line,68}]},{application_master,start_it_old,4,[{file,"applicat                                   ion_master.erl"},{line,273}]}]}}}}}}},[{lorawan_app,start,0,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_app.erl"},{li                                   ne,16}]},{init,start_em,1,[]},{init,do_boot,3,[]}]}}
> init terminating in do_boot ({{badmatch,{error,{lorawan_server,{_}}}},[{lorawan_app,start,0,[{_},{_}]},{init,start_em,1,[]},{init,do_boot,3,[]}]})
> 
> Crash dump is being written to: erl_crash.dump...done

gotthardp, 2017-08-16:
eaddrinuse. Some other program. Possibly another instance of the server is using the tcp/udp port needed.

oyj0594, 2017-08-16:
thnks. i understand what's going on. :)

closed 2017-08-16 by oyj0594