gotthardp/lorawan-server #187: Erorr which cause gateway freez

Kralg, 2017-08-23:
Hello,

We were running gateway for a while and once it got frozen. All LED's are shining and command line shows this.
08:50:17.237 [info] JOIN REQUEST 70B3D57EF000425C 0000000000000010 -> 4486AFED
08:50:17.237 [debug] Join-Accept in RX1: {0,0,869.525}
09:24:51.551 [error] HTTP 404 /links?_page=1&_perPage=7&_sortDir=ASC&_sortField=devaddr
09:24:52.930 [error] HTTP 404 /links/11223344?

and after the restart once gateway connects to the server:
09:45:37.777 [error] HTTP 404 /nodes/12?
09:49:16.387 [error] HTTP 404 /nodes/12?

Does it mean that there are some commands over HTTP? This is interesting as I am not using any such commands.

gotthardp, 2017-08-23:
Did the server freeze, or the entire gateway OS? Was there a browser connected to the gateway somewhere?

Kralg, 2017-08-23:
Well, that is a good question. I was able to navigate in webpage and raspbian was running, just Concentrator board was statically shining. Once server was restarted, led's changed to normal state. It could be that browser was connected over the local network.

gotthardp, 2017-08-23:
It's hard to guess what happened. I cannot reproduce this behaviour.

closed 2017-08-25 by gotthardp