gotthardp/lorawan-server #188: User interface shows unconfirmed messages.

gjoosen, 2017-08-23:
Hello,
I've added a Node in the UI, and I do receive data, but the UI shows the messages are unconfirmed. 
It seems that the packet forwarder does receive acks. I've added a screenshot of the UI as well as the packet forwarder log. Is this a configuration problem?
[image]
[image]

gotthardp, 2017-08-23:
Hello. Unconfirmed means the device didn't request an ACK. You need to convince your device to send confirmed uplinks.

gjoosen, 2017-08-23:
So, in this case the server didn't send an acknowledge to the node?

gotthardp, 2017-08-23:
No it didn't. Because the node didn't request it.

gotthardp, 2017-09-08:
The user interface included a bug, so confirmed messages were shown as unconfirmed (and vice versa). Will be fixed in 0.4.11.

closed 2017-09-08 by gotthardp