gotthardp/lorawan-server #192: Configure and observe the maximum DR/SF per device

gotthardp, 2017-09-05:
- if ADR is enabled in the device the server assumes the device will obey its commands and if the device replies with a failure it tries again and again forever
 - the maximum supported datarate is hardcoded and cannot be modified for each device individually

stefanobimbo, 2017-09-22:
I have another strange behaviour with ADR.
Server is configured with ADR ON for a given node. It tries to send an ADR request that the node rejects. Now I switch off the node and I think it is switched off before the ADR reply was sent by node (I'm not sure of that). Then I switch the node on again after some time. The node starts sending an uplink with framecnt=0 (I use join=ABP). I configured the node in the server DB with FCntCheck="Reset on zero".
The server immediately after the reception of the uplink message, sends an ADR request to the node, but nothing is logged to debug.log. I think the ADR command is queued to the downlink queue of the node and is not deleted when the frame 0 is received (as I may aspect). The same happens even if I disable ADR and repeat the sequence

gotthardp, 2017-09-22:
Could you share that log with me, please? There is no ADR command queuing, so the reason must be different.

Resetting a device:
 * does reset the effective (used) ADR
 * does _not_ reset the desired ADR

So, if a device was using a non-default ADR and then reboots, the server tries to switch it back to the desired value (even though it was rejected). Is this what are you experiencing?

stefanobimbo, 2017-09-23:
There's nothing in the server log that I can see, but I see the uplink and downlink messages in the gateway log (even if I can't see data in clear form in the gateway log, of course). [I can send the gateway log to you if you are interested in].
What do you mean with "non-default ADR"?. Desired ADR value is the value that ADR computes before the node was switched off? If the ADR is set to OFF, does the server ignores this setting and tries to send the ADR request command anyway?
The node starts with data rate=2. Can it be the problem?

gotthardp, 2017-09-23:
* There must be a `debug.log` somewhere. I'd be interested to see that file. Gateway log doesn't help me much.
 * LoRaWAN standard defines what ADR settings shall be used after reset. I called that 'default'.
 * If ADR is on, the server calculates the 'desired ADR' after receiving each frame.
 * If ADR is off, the server sends an ADR command everytime the 'used ADR' (default ADR after reset) differs from the 'desired ADR'.
 * If the node starts with data rate=2, then you confuse the server as it thinks the device starts with rate=0, which is the default one; this could definitely cause some unexpected behaviour

gotthardp, 2018-01-11:
Implemented for the future 0.5.0 release.

closed 2018-01-11 by gotthardp