gotthardp/lorawan-server #194: packet with fcnt 0 not handled

lignode, 2017-09-07:
after each join accept, when device send an confirmed packet with fcnt 0, the server just ignore it.  it looks like after each join accept, the server reset fcntup to 0. then when it receive a fcnt 0 packet, server process the packet as an retransmitted packet because of the same fcnt 0.
is this behaviour correct? i'm very confused, because semtech test case send their first packet after joining with fcnt 0 instead of 1.

gotthardp, 2017-09-07:
I am confused too, but you may be right. I will double check with other implementations.

qingchuwudi, 2017-09-07:
Yeah,it seems that `0` while be updated to `1` by mote program in the first frame.

[LoRaMac-node](https://github.com/Lora-net/LoRaMac-node)

gotthardp, 2017-09-07:
And does PrepareFrame send 0 or 1 in the first uplink? I thought I did test it and it sent 1.

qingchuwudi, 2017-09-07:
yeah, it does send `1`.

gotthardp, 2017-09-07:
I don't want to change it if the reference node firmware sends 1. Could someone prove that sending 1 is wrong?

lignode, 2017-09-08:
there is a change in reference node code. please look at commit  075e4fc2c62863c88849a312d3163e4beef874f8  of https://github.com/Lora-net/LoRaMac-node .
in earlier code, the UpLinkCounter is initialized to 1, after that, 0.
seems like Semtech not very clear on their own spec :(

qingchuwudi, 2017-09-08:
My server is using `1`, and I'm waiting for the new LoRaWAN spec. Although, node code has bean initialized with `0`.

gotthardp, 2017-09-28:
Other users suffer from this problem too, so I made a fix to allow a device after join to start with 0 or 1.

gotthardp, 2017-09-28:
Release in v0.4.11

closed 2017-09-28 by gotthardp