gotthardp/lorawan-server #198: gotthardp lorwan server on Multitech conduit mlinux

pdl86, 2017-09-13:
Hello 
does anyone succeed to install the lorawan-server on a Multitech conduit mLinux gateway ? 
we need to build simple stand-alone network ( but with possibility later one to add two or three gateways as slaves), integrtaed MQTT connector  and with easy to use lorawan-server ( thank to petr GUI) .

any feedback will be appreciated.
regards
Philippe

gotthardp, 2017-09-13:
Not yet, but working on this.

pdl86, 2017-09-13:
Hi Petr

good news.

Any expected date ?

may I help ( testing, check , ...) ?

regards

philippe

gotthardp, 2017-09-21:
Hello. I finally reached some success. It is documented [here](https://github.com/gotthardp/lorawan-server/blob/master/doc/Embedded.md#multitech-mlinux). May need some testing and tuning though. Feel free to suggest updates.

pdl86, 2017-09-26:
Hello

thanks for teh info , we will give a try next week.

closed 2017-09-27 by gotthardp