gotthardp/lorawan-server #201: Can not create Handler

pdl86, 2017-09-18:
Hello 
I'm trying to create an handler , but get a "Status Change error" systematically

What's strange  : 
1°  Format : whatever is chosen ( Json, raw data, Web form) , the "ticker valid /invalid" shows both ( X and V)
2 uplink  fields : I suspect that "rxq" has to be chosen for parsing ? 
 ( by the way , ,where are coming from these fields value ? )
3 Parse uplink and Build downlink are almost "standard" ( derived from the doc) 
4) the connector is existing , but the "ticker valid /invalid" shows both ( X and V)

what could  be wrong ? 

[image]

thanks in advance 
philippe

pdl86, 2017-09-18:
Finally I've found  the error, withing the Parse Uplink syntax,  neither Long nor AltGPS , in the second member were  existing, and the order of assignation was inverted.
sorry for the distrubance

closed 2017-09-18 by pdl86

gotthardp, 2017-09-18:
True. While at this-- please note that the variables in the `<< >>` brackets shall start with a capital letter. The first `ts` doesn't (if I see correctly).

pdl86, 2017-09-18:
Petr
thanks for the tip: it will probably save me some serious time !
regards
philippe