gotthardp/lorawan-server #204: Docker Image doesn't work for the Raspberry Pi 3

kekbait, 2017-09-21:
I spent some time creating a Docker image working with the Rasberry Pi3 (Arm7 Architecture).
Maybe you could include it somewhere to help people in the same situation ?

Thanks a lot for your awesome work !

```
FROM tedsluis/raspbian.minimal.arm7
RUN apt-get update && apt-get install -y wget git npm make

RUN apt-get install -y libssl-dev
RUN apt-get install -y ncurses-dev
RUN  wget http://www.erlang.org/download/otp_src_20.0.tar.gz
RUN  tar -xzvf otp_src_20.0.tar.gz
RUN  cd otp_src_20.0/ && ./configure && make && make install

RUN rm otp_src_20.0.tar.gz
RUN rm -R otp_src_20.0/
RUN mkdir lorawan-server
RUN wget https://github.com/gotthardp/lorawan-server/releases/download/v0.4.10/lorawan-server-0.4.10.tar.gz
RUN mv lorawan-server-0.4.10.tar.gz lorawan-server/
RUN cd lorawan-server && tar -zxvf lorawan-server-0.4.10.tar.gz

 # volume for the mnesia database and logs
RUN mkdir -p /storage
VOLUME /storage
# data from port_forwarders
EXPOSE 1680/udp
# admin interface
EXPOSE 8080/tcp 
ENV LORAWAN_HOME=/storage 
WORKDIR /lorawan-server
CMD bin/lorawan-serve

```

gotthardp, 2017-09-21:
Thank you very much. I added it to scripts/bitbake. Do you want to write some instructions on how to use it, please?

kekbait, 2017-10-06:
Sure.
You just need to copy-paste the content from my previous post in a file named "Dockerfile" and run those two commands :

`$ docker build -t gotthardp/lorawan-server .`

`$ docker run  --restart unless-stopped --detach --name lorawan --hostname lorawan --volume /root/storage:/storage --publish 8081:8080/tcp --publish 1680:1680/udp --env GOOGLE_MAPS_KEY=xxxxxxxxxxxxx-xxxxxxxxxxxxxxxxxxxxxxxxx gotthardp/lorawan-server`

closed 2017-10-06 by gotthardp