gotthardp/lorawan-server #209: How to create a Lorawan app

sokolbiba, 2017-09-28:
Hi,
I need to know how can I create an application that runs on a lorawan-server. I am working with ic 880a module and what I need to do is not to use the ttn (the things network) server but my own one.So far I have followed the steps to install the lorawan server and erlang (is there another way not to use erlang).
I am stucked in the step of creating a simple app that forwards the received packets in the gateway to my own lorawan-server.
Thank you in advance. :)

gotthardp, 2017-09-28:
Hello. The server is receiving frames and you want to process them *or* your gateway is not forwarding packets to your gateway?

To build a custom app you can implement it in the server directly (my preferred way, but you'd need to learn the erlang language). Alternatively you can build your own app in any language you like and receive the frames over MQTT or WebSocket. The latter is perhaps quicker to do, but less robust and slower.

gotthardp, 2017-09-28:
What kind of gateway you have? Did you configure it as described in the documentation: https://github.com/gotthardp/lorawan-server/blob/master/doc/Installation.md#configuration-of-the-packet_forwarder?

sokolbiba, 2017-09-28:
The gateway I am using is ic880a (picture below). I have followed the steps but could you please telling me if the file  global_conf.json that I need to modify is the one found in /opt/ttn-gateway/packet_forwarder/poly_pkt_fwd.

[image]

gotthardp, 2017-09-28:
Yes, the poly_pkt_fwd's global_conf.json or local_conf.json should be the right one. Does it specify the port 58989? You need to modify the server a port.

sokolbiba, 2017-09-28:
In the global_conf.json I have the configuration as depicted in the picture below where the ip 192.168.0.207 is the server ip address. 
[image]

gotthardp, 2017-09-28:
Good. Remove the second `{ ... }` record and keep the first serv only. Then, change serv_port_down to 1680 (same as _up) and modify the server_address to where your server is running.

sokolbiba, 2017-09-28:
Now it seems that it is working. I can see that the ping is working and the gatway is forwarding the packets to my server. What I would like to do as the next step is: how can I extract the information in the packet in a readable form for human kind. Can you help me with that? 
[image]

sokolbiba, 2017-09-28:
I am not sure about that. How can I access the web-admin of the server ? I installed server on a kalilinux machine but I do not know how to access it ? Do you have any suggestion ?

sokolbiba, 2017-09-28:
I guess that you are talking about this. But I am not sure why I do not see any data there :/ 
[image]

gotthardp, 2017-09-28:
I suspect there will be some errors in the "Events" section. You need to configure at least one gateway and at least one device/node.

sokolbiba, 2017-09-29:
Thank you very much. Seems that now I have at least one node transmitting Hello World! Thanks @gotthardp

gotthardp, 2017-09-29:
Hello. I am not sure what application you are building, but you may have a look at https://github.com/gotthardp/lorawan-server/blob/master/src/lorawan_application_microchip_mote.erl, which is a full implementation of a very small demo app in Erlang.

sokolbiba, 2017-09-29:
Hi.
What I want to do next is to stop the gatway sending information to ttn network and only sending the information to my own server. I tried the command 'sudo service lorawan-server stop' but this stoped it just for a while and it starts again.  How can I do that ?

gotthardp, 2017-09-29:
But you already did configure that, right? (When you removed the second `{ ... }`, see above.)

sokolbiba, 2017-09-29:
Yes I removed the second {....} but I can see that in the ttn network my gateway is online even that I am not sending information to the ttn. What I need to do is, lets say to deattach my gateway from ttn so I wont be able to access it from ttn.

gotthardp, 2017-09-29:
I don't know how this detachment works, unfortunately. All I can suggest is to try searching for other global.conf or related local.conf files.

sokolbiba, 2017-09-29:
Thank you for all.

gotthardp, 2017-09-29:
Did it work? :) May I close the issue?

sokolbiba, 2017-09-29:
I will write a new comment tomorrow because I am at home tonight to give it e try. So I will let you know tomorrow or in Monday.

sokolbiba, 2017-10-02:
Hello friend, 
I am trying to deploy the OTAA method. I am using the arduino library https://github.com/matthijskooijman/arduino-lmic . In the server side I can see that the RF packets arrived and the server is able to generate dev_addr, netskey, appskey. The problem is that I am not able to receive the generated data to arduino node. Below you will find a screen shot of the message from the server and the from the arduino console. Do you have any idea on that ? 
[image]
[image]

sokolbiba, 2017-10-02:
[image]

gotthardp, 2017-10-02:
Yeah. The LMIC devices have troubles with RX1 window. You may want to set the *TX Window* of your Device or Node to *RX2*.

gotthardp, 2017-10-03:
I don't think so. That timestamp is not supposed to denote any date; it's a number of microseconds since some undefined moment in time.

sokolbiba, 2017-10-03:
Still I am not able to receive the nwskey and appskey and devv_addr. Would you miind to suggest me an arduino example (that you have tested yourself)?

gotthardp, 2017-10-03:
I don't have any, but you may ask others on the google group related to this server.

sokolbiba, 2017-10-04:
Hi friend.
I see that lorawan-server is able to store some data coming from nodes in the network and I would like to know if it is possible to access this kind of database where all the information is stored. Is it possible for me to access this database and where can I find the it? Thanks :)

gotthardp, 2017-10-04:
See the JSON API https://github.com/gotthardp/lorawan-server/blob/master/doc/Administration.md
The frames are stored in the `rxframes` table.

gotthardp, 2017-10-07:
May I close this issue?

sokolbiba, 2017-10-08:
Thank you friend. Go ahead and close this. I really appreciate your help :)

closed 2017-10-09 by gotthardp