gotthardp/lorawan-server #210: Error messages in log

javierpedrido, 2017-09-29:
Hello,

I'm seeing this error logged every time a some message comes in.

2017-09-28 22:50:18.375 [error] <0.289.0>@lorawan_application_backend:data_to_fields:222 Fun failed error:function_clause

But everything works fine, the data is parsed and is sent to mqtt.

My parse function is:

fun(_Port, <<Count, Data:Count/binary>>) ->
  #{data => binary_to_list(Data)}
end.

Javier.

gotthardp, 2017-09-29:
But the message sent doesn't contain the parsed `fields` attribute, right? (See https://github.com/gotthardp/lorawan-server/blob/master/doc/Backends.md#parse-uplink).

You're getting the message because the application data does not correspond to the binary pattern ` <<Count, Data:Count/binary>>`. I guess they may not start with count.

javierpedrido, 2017-09-29:
Hi, you are wright, I just copied the example in the docs and didn't think much about it (New to Erlang). Thanks!

gotthardp, 2017-09-29:
Cool :)

closed 2017-09-29 by gotthardp