gotthardp/lorawan-server #211: Changing ADR parameter messes up up/downlink counters

javierpedrido, 2017-10-01:
Hi,

I think I found a bug. I have one ABP node with ACks turned on and configured as manual ADR. If I stay in the ADR page while the node sends data, then I change the DR and hit save, node ACKs reception starts failing. The problem is that saving the DR change also saves the counter values from the moment I entered to the ADR page and that out synchs the counters causing the ACKs reception to fail in the node. If I always refresh the page before doing the DR change and saving, everything works OK.

Regards, Javier

gotthardp, 2017-10-01:
Hi. Thanks for reporting this. I happened to me as well, but I forgot about it. Will try to fix that.

gotthardp, 2017-10-02:
All I can do is to block all web-updates if the data changed in between. In real life it will not be that likely situation, what do you think?

javierpedrido, 2017-10-02:
Hi, I think the situation is very common when testing the software and that an informative message asking for a refresh is OK. It took me sometime to realize what was going on.

gotthardp, 2017-10-03:
I finally fixed this. Thanks for reporting this issue. Even though I was aware of it; you made me to fix it. ;-)

closed 2017-10-03 by gotthardp

javierpedrido, 2017-10-03:
:), thanks for the fix, I don't know what the decay problem is, but it seems to be working OK.