gotthardp/lorawan-server #213: issue on Timestamp

pdl86, 2017-10-05:
Hello 
when we process the frames received from Gotthardp via a mqtt connector, the timestamp field ( tmst) contains strange value , not related to the real time stamp ( as seen by the lorawan server 
exemple 
frame received in gotthardp at 2017-10-05 19:39:16  (france)  ( normally ts should be: 1507225156)
received  then via mosquitto broker with tmst = 1131506996 , Wed, 09 Nov 2005 03:29:56 GMT ???

any clue ? 
regards
Philippe

gotthardp, 2017-10-05:
Hello,
the [documentation](https://github.com/gotthardp/lorawan-server/blob/master/doc/JSON.md) describes tmst as an "internal timestamp". It's not a standard timestamp; it's a number of microseconds since some unknown moment. This moment is gateway specific.

pdl86, 2017-10-05:
ok
is there a way to obtain the Lorena network timestamp ?   , ideally the
time of emission by a device ?
regards

philippe

gotthardp, 2017-10-05:
the gateway sends timestamps only when it has GPS. does your gateway have gps?

pdl86, 2017-10-05:
yes and the devices are gps trakers.

 by the way thanks for the support on multitech conduit. it's working quite
well .
regards

gotthardp, 2017-10-05:
If the gateway has GPS then you should see rxq.time, do you? If there is no rxq.time, then I suspect the GPS might not be working properly.

pdl86, 2017-10-06:
Hello 
we do not have rxq.time in the frame sent by Lorawan.server to mosquitto , the gps is working fine ( in parallel , we query gpsd to get the position every minuet, moreover, the gps is used to sync the time, and it's ok) 

for example , this is what with get via mqqt : 
{"data":"0047319100008E9728A0018E","devaddr":"0024BBF5","fcnt":23,"group":"FS_tracker","port":1,"rxq":{"codr":"4/5","datr":"SF12BW125","freq":868.5,"lsnr":6.5,"rssi":-35,"tmst":2835187340}}

so any any help ill be appreciated.
regards
Philippe

gotthardp, 2017-10-06:
For start you can add `datetime` to "Uplink Fields" in the setting of your connector. It's the server time, but at least something for now.

Then, would you be able to check the gateway to server UDP communication and send some `rxpk` structure the gateway sent to the server?

pdl86, 2017-10-06:
Hello

>
> For start you can add |datetime| to "Uplink Fields" in the setting of
> your connector. It's the server time, but at least something for now.
yes, it wokrs ( in fact we did already that by extracting the local
gaetway time and adding it to the frame send to application)
>
> Then, would you be able to check the gateway to server UDP
> communication and send some |rxpk| structure the gateway sent to the
> server?
this a raw UDP packet ( sent by lora-wan server) as captured

E$@@_[9~$:V
0v1/gotthardp{"data":"0047329B00008EC32EE00178","datetime":"2017-10-06T11:28:43Z","devaddr":"00254FD3","fcnt":11,"group":"FS_tracker","port":1,"rxq":{"codr":"4/5","datr":"SF12BW125","freq":868.3,"lsnr":9.0,"rssi":-21,"tmst":2001624364}}

no trace of rxq.time

regards
philippe

gotthardp, 2017-10-06:
But this is UDP from the server. Could you get UDP sent by the gateway, please?

pdl86, 2017-10-06:
Hi

what do you mean by the gateway ? ( as the lorawan server is running on
the multitech gaetway !) : the packet forwarder ?

here is a capture of some traffic ( pcap file)

I see a timestamp ( tmst) in the packets  , as weel as date/time in the
"stat" packet .

any clue ?

regards

Philippe

gotthardp, 2017-10-06:
I cannot see the pcap file. Could you send it to my email, please? petr.gotthard@centrum.cz

pdl86, 2017-10-06:
Hi Petr

the file is ( was) attached .

In parallel I upload it on the git server

regards

Philippe

Hi

what do you mean by the gateway ? ( as the lorawan server is running on
the multitech gaetway !) : the packet forwarder ?

here is a capture of some traffic ( pcap file)

I see a timestamp ( tmst) in the packets  , as weel as date/time in the
"stat" packet .

any clue ?

regards

Philippe

pdl86, 2017-10-06:
[packet_forwarder.zip](https://github.com/gotthardp/lorawan-server/files/1362827/packet_forwarder.zip)

gotthardp, 2017-10-06:
In the log you can see "stat" packets with time, but these are not related to any frames. Then you can see the rxpk frames, but without any timestamp (only the tmst number). The packet_forwarder (from some reason) does not send any timestamp info to the server, so the server cannot forward it to MQTT.

pdl86, 2017-10-06:
Hi

OK ,
any clue ? a mis configuration ?
we are using te "standard" packet forwarder of Multitech ,  
I do not see really a way to configure ( or not) the GPS clok source ?
any hints
regards
Philippe

gotthardp, 2017-10-06:
The packet_forwarder comes in 2 flavours: basic and gps. Perhaps your gateway does not run the gps flavour. BTW, my Multitech Conduit has no GPS, your really does?

pdl86, 2017-10-06:
hi
yes the newest versions (since July ) can have a gnss option.
I will see with milultitech support and will let you know.
By the way, on another topic where do you stand with downlink ?
regards
philippe

gotthardp, 2017-10-06:
OK, thanks.
What kind of downlink you mean?

pdl86, 2017-10-07:
Hi

I was referrng to "HTTP backend for downlinks #171"

regards

Philippe

gotthardp, 2017-10-07:
I have it on my todo list with no target date. I have to focus on other priorities for now.

pdl86, 2017-10-07:
no worries as we can adress that via websocket or Mqtt and some sniffing of
needed

regards
Philippe
philipe

gotthardp, 2017-10-07:
May I close this issue now?

pdl86, 2017-10-07:
if you can better to wait the feedback of multi tech support
thanks
reds
philippe

gotthardp, 2017-10-07:
Sure! :)

pdl86, 2017-10-12:
Hi
after several exchanges with Multitech, as per their statement, the mLinux version is a "developpement" version, and as such , some features may not be fully implemented or tested . the rxpk.time attribute beeing one of them . !!!
they  suggest to use a utility : gpspipe process to "pipe" the GPS time into the packet_forwarder, and admit that their engineer were not able to have it working in a stable way !
they suggest to move on the AEP version, aimed at developping application, but then stopping to use Gotthardp lorawan server, unless to make the effort to integrate  it.
So as a result , we will spend some (mimited) time to sse if we can "fix" the gpspipe usage , otherwise implement the timestamping as part of the data payload from the device .

so I think you can close the issue 
regards
Philippe

gotthardp, 2017-10-12:
Thanks for the info.

closed 2017-10-12 by gotthardp