gotthardp/lorawan-server #214: Implementing Handlers and Connectors

sokolbiba, 2017-10-06:
Hello everyone,
Actually I am lorawan server to create a private lora network. My end-device (I only have one device for now) is transmitting some hello world payload and everything seems to be a great job with the server. I would like to know more how can I implement/use handlers/connectors to get this payload (hello world) and to send it to another server where I have created my website. Is that possible using the lorawan-server?

gotthardp, 2017-10-06:
Hello. Did you read
https://github.com/gotthardp/lorawan-server/blob/master/doc/Integration.md
?

sokolbiba, 2017-10-08:
I read that part but seems a bit messy to me. I am not really friendly with mqtt servers. I see that I should learn some basic about mqtt servers.  Can you suggest a faster guide for that ?

altishchenko, 2017-10-08:
Hello,
It all depends on the service you are trying to connect to. If this is a public cloud service, like AWS, Azure, Adafruit - you may find a helpful how-to in their guide pages. If this is your own service/MQTT broker:
1. As a client - you publish to a topic /foo/bar/baz...
2. If you need, you listen for topic(s), say for commands back to your motes - if it is a controllable mote  (like a wall outlet - on/off). You can listen for one topic only or a bunch of topics selected by wildcards /floor/room/* - have a look at emqttc docs at github (that is what the server uses in the background).
And you can always find the latest MQTT info on mqtt.org site.

altishchenko, 2017-10-08:
Write me at atishch@gmail.com, if you still have any questions concerning integration.

gotthardp, 2017-10-09:
Also, there is a nice [Get Started Page](http://www.rabbitmq.com/getstarted.html) for RabbitMQ, which also has MQTT plug-in.

sokolbiba, 2017-10-09:
@altishchenko  thank you for your suggestion. I really appreciate it. 
@gotthardp  I successfully  installed the rabitmq-server using the tutorials from the offical page. For the moment I am using http://www.rabbitmq.com/tutorials/tutorial-one-php.html example and is running successfully with the examples send.php and receive.php.  Know that I have rabbitmq-server up and running I am facing issues when trying to implement connectors and handlers. Below you will find two screenshots on how I have configured the connectors and handlers. Do you have any idea what is going wrong with the configuration made by me ?  I am not sure if the lorawan-server is able to send data to rabbitmq using connectors, handlers or both of them ? 
[image]
[image]

gotthardp, 2017-10-09:
Please make sure you have the MQTT plugin-enabled (https://www.rabbitmq.com/mqtt.html) It's not enabled by default and the RabbitMQ examples are using different protocol.

Common mistake is that Device/Node is not configured to use the connector. You have to:
 * Set Device/Node Application to *backend*
 * Set Group to the name of your handler, i.e. `test` in your case

If you did all this and it still doesn't work, could you please have a look in the lorawan-server logs if there is not any error?

sokolbiba, 2017-10-10:
Hi again,
After your comment I enabled the mqtt plugin but still I am not able to connect to the Rabbitmq-server. Please have a look at two screenshots to get an idea on how I am configuring the handler and the connector. Also you will find a copy of debug.log from lorawan server. I do not know why it is not working when I put everything as said in the documentation. In this post I left the published topic,Subscribe and Received topic fields empty just to check if there is any difference in debug file. 
[image]
[image]
[debug.log](https://github.com/gotthardp/lorawan-server/files/1371026/debug.log)

sokolbiba, 2017-10-10:
I figured out that the port was not the right one. Now the port of the connector is on 1883 and the debug.log changed somehow. Below you will find the debug.log file and a  file.config for the rabbitmq-server. Please have a look and let me know. Thanks :D 
[debug.log](https://github.com/gotthardp/lorawan-server/files/1371312/debug.log)
[image]

gotthardp, 2017-10-10:
Could it be because you didn't enter the *Authentication' credentials? The user/pass is guest/guest.

sokolbiba, 2017-10-10:
[image]
Probably yes, I figured out that was some port and credentials wrong. Know it is working with rabbitmq-server but I do not understand that why the backend application is showed as a consumer in the rabbitmq management UI. Please take a look at the screenshots of the debug.log and the rabbitmq UI.  The queue generated by backend application seems to be a random queue named, smth like subscription-emqttc_kali_3d723b1bba85d9c5d225qos1.
[debug.log](https://github.com/gotthardp/lorawan-server/files/1371704/debug.log)

gotthardp, 2017-10-10:
For uplink this looks good. the term "consumer" is an MQTT terminology. This is OK.

sokolbiba, 2017-10-10:
I am trying to get the data from the queue by clicking on Get Messages and there is no data in the queue. Would you mind if I give you the IP of the lorawanserver to check if I am configuring it in the right way or not and send me a feedback ?

gotthardp, 2017-10-10:
No problem :) Will you give me access to RabbitMQ too?

sokolbiba, 2017-10-10:
Please send me an email address and I will send you the IP and credentials in your email.

gotthardp, 2017-10-10:
petr.gotthard@centrum.cz

gotthardp, 2017-10-10:
I will better document the queue/exchange binding.

closed 2017-10-10 by gotthardp