gotthardp/lorawan-server #215: Bug when turning on ADR

javierpedrido, 2017-10-10:
Hi,

I think found a bug, I had one OTA node that was created with ADR off, it sent some data during two days, then it was reseted and after that it rejoined and sent more data. Then I decided to set ADR to Manual in that node and I couldn't do that because I got a "Data has changed. Please reload and repeat the action." message. The same behavior has observed if I changed the ADR config from the node or from the device.

Regards, Javier

gotthardp, 2017-10-10:
This is a new feature to prevent database corruption. Did you reload the page and retry?

javierpedrido, 2017-10-10:
Yes, I know about it, but even after refreshing the message keeps appearing and i can't save the change.

gotthardp, 2017-10-10:
Could the device be sending data very frequently? If you refresh and the device sends another frame before saving the message will appear again.

javierpedrido, 2017-10-10:
No, it is sending data sending every half hour. This device was created with an old version of the server, could that be the cause?

gotthardp, 2017-10-10:
It shouldn't be the case. Does the Node indicate some warning (yellow/red exclamation mark) or not?

javierpedrido, 2017-10-10:
No, there's no warning. The u/l links counters were resetted to zero during the node reset, could that be cause?

gotthardp, 2017-10-10:
The server basically calculates a checksum what was on the page and rejects an update if that checksum changed. Could there be something in the DB that changed while you were editing the Node?

javierpedrido, 2017-10-10:
No, there's nothing using the db apart from the server.

gotthardp, 2017-10-10:
Then it's a mystery. Would it be possible to send me trace of server traffic (both gateway to server as well as the HTTP traffic)?

javierpedrido, 2017-10-10:
I will try to do it tonight, would you like a copy of the db?

gotthardp, 2017-10-10:
No. Just the .PCAP file(s) and perhaps the debug.log from the same time.

javierpedrido, 2017-10-10:
This gets logged in debug.log:

2017-10-10 12:23:20.076 [warning] <0.469.0> server {http_error,{412,"/nodes/02FD75D0"}}

javierpedrido, 2017-10-10:
One other thing, that's not the normal setup is that I'm proxying 443 to localhost:8080 using a nginx server.

javierpedrido, 2017-10-10:
The proxy is the problem (I don't known why), when I access the server directly it works OK.

gotthardp, 2017-10-10:
This is possible; I didn't test with nginx proxy. I keep this issue open, but as a low prio, OK?

javierpedrido, 2017-10-10:
Yes no problem, I'm not in a hurry, thanks!

gotthardp, 2017-10-10:
I did a little update, which may have fixed that (or not). Would you please clear the cache of your proxy and then try again?

javierpedrido, 2017-10-10:
It didn't work, I'm using this configuration by the way:

location / {
        proxy_pass http://localhost:8080;
        proxy_set_header Host $host;
        proxy_set_header X-Real-IP $remote_addr;
        proxy_http_version 1.1;
        proxy_set_header Upgrade $http_upgrade;
        proxy_set_header Connection "upgrade";
        if ($scheme = http) {
            return 301 https://$server_name$request_uri;
        }
}

javierpedrido, 2017-10-10:
The problem is not caused by using a proxy, it's because I've turned gzip compression globally on. That's causing the problem.

gotthardp, 2017-10-10:
gzip? I am not a proxy expert, but I'd say this is not a server bug then.

javierpedrido, 2017-10-10:
I don't know why it's not working, it seems like there's no bug in your app. I could try not to gzip application/json types, but I don't need the lorawan server app to be compressed, I just need it in my web site (but I turned it on globally).

Sorry for the waste of time.

closed 2017-10-10 by gotthardp

javierpedrido, 2017-10-10:
By the way, not compressing application/json mime types, "solved" the problem.

altishchenko, 2017-10-10:
is this piece of code the correct way to express comparison? :
if ($scheme = http) {
return 301 https://$server_name$request_uri;
}

IMHO, should be == or != (Most probably) -- correct me if I am wrong...
Just given the code above, it must always return 301... Assignment to http,
which is non-zero value, which is 'True' by all ifs, will execute 301
return every time this bit is passed, not even caring what the _real_
scheme was in the request.

С уважением,
Александр Тищенко
+7 (916) 704-6157
atishch@gmail.com
=====

javierpedrido, 2017-10-10:
The syntax is ok, yes, the server cannot be accessed via http, all requests are redirected to https that is proxied to localhost:8080.

altishchenko, 2017-10-10:
You missed the point. Usually the construct is if ($a == b) or if ($a !=
b). Single '=' is an assignment in any programming language I know of (and
I know quite a number). Basically, what that 'if' line is stating is:
1. $scheme = http (So, now the $scheme variable IS set to be 'http').
2. Is 'http' itself is False? (Null, None, etc...) - No? Then go and send
301.
And just http by itself (as a constant or object/module) will never be
False, so you will always get 301.

Just try hand querying your server with telnet or get a tcpdump/wireshark
trace to witness that.

Or, tell me what kind of a programming language uses a single '=' as a
comparison? I only know that basic and Fortran did that, don't truly
remember if Ada did that too or it already used doubling...

Or just try putting '==' instead of '=' in that very 'if' and check the
results.

С уважением,
Александр Тищенко
+7 (916) 704-6157
atishch@gmail.com
=====

javierpedrido, 2017-10-10:
altishchenko, add nginx configuration files to your list, the syntax is correct :)

altishchenko, 2017-10-10:
Aha, smoked a bit of that right now. Luckily, never had to touch it. And
there is another question to it... Should it not then be $scheme = "http"
(Mind the quotes) or http is a sort of handler/object/module name variable?

(Petr, I hope you don't regard this as spam?)

С уважением,
Александр Тищенко
+7 (916) 704-6157
atishch@gmail.com
=====

altishchenko, 2017-10-11:
> Quotes are not needed for a value like http

Hmm, so it is a pre-defined object?
Just curious, can you drop a few debug prints around this code just to see how it is branching?

С уважением,
Александр Тищенко
тел: +7 (916) 704-61-57
atishch@gmail.com