gotthardp/lorawan-server #216: Change time from UTC to local time of the Raspberry

sokolbiba, 2017-10-11:
Please have a look at the Last Rx field and the time in the graph. Why they do not correspond to each other. I think that the time in the last Rx field corresponds to the local time of the  and the time in the graph is the time in UTC format. Is it possible to change UTC to local time ? 
[image]

gotthardp, 2017-10-11:
I don't know how :) I use the [Dates and Times Using the Date String Representation](https://developers.google.com/chart/interactive/docs/datesandtimes), which are probably(?) in UTC and the fully rely on google charts. I believe it's a limitation of Google Charts, but I am not sure.

altishchenko, 2017-10-12:
I'll have a look into this, as soon as the guys at ESL (@bszaf) will fix the release build process for RPi.

gotthardp, 2017-10-12:
Thanks, but please note this is not a RPi issue. This is a generic problem with all graphs that have a timestamp as the X-axis-- it shows UTC time instead of the local time.

altishchenko, 2017-10-12:
I do understand that. But, for clarity and purity I’d like to do that on RPi, because it has its own gotchas concerning localization.

Cheers,
Alex

gotthardp, 2018-08-25:
This depends on https://github.com/google/gwt-charts/issues/42