gotthardp/lorawan-server #218: I'm not receiving any data in Adafruit IO

Viswanath95, 2017-10-12:
I'm trying to send data from lorawan server to Adafruit IO.I followed the Integration Instruction given by this site: https://github.com/gotthardp/lorawan-server/blob/master/doc/Integration.md
But still now i didn't receive any data in Adafruit IO
I have a doubt on subscribe part my lorawan server user name is : Admin and my all data is in Received frames so in lorawan server on subscribe part as per the given Instructions I wrote: 
      admin/feeds/Received Frames Is this is the correct way of creating a subscribe part?
And i created feeds and Dashboard in Adafruit IO but i don't know why the data is not shown In Adafruit IO
                                                                                                                   Thanks in Advance

altishchenko, 2017-10-12:
Hello Viswanath,
There is a cure for that. The reason you are not receiving anything is that io.adafruit.com <http://io.adafruit.com/> expects a valid JSON payload in your MQTT publish request. We should update the Integration.md to reflect that. Their JSON format expects a payload of the form { «value»: 123456 } for the particular feed.

1. Create a connector with:
URI: mqtt://io.adafruit.com <mqtt://io.adafruit.com>
Published Topic: yourlogin/feeds/yourfeed

2. Build a handler with Format field set to ‘Raw Data’, your Adafruit connector as Connector
AND add a Parse Uplink fun, something like the one below:

fun(_Port, <<_Humid:16, Temp0:16, _Csum>>) ->
TSign = Temp0 band 16#8000,
TVal = Temp0 band 16#7FFF,
case TSign of
  0 -> Temp = TVal / 10;
  _ -> Temp = -(TVal / 10)
end,
[H|_] = io_lib:format("~w", [Temp]),
 <<"{\"value\":", (list_to_binary(H))/bytes, "}">>
end.

The first line here parses the binary data from the sensor: it sends five bytes humidity, temperature and checksum. I am ignoring humidity and checksum, hence the _ in front of the variable names. The temperature itself is sent in tenths of a degree, that is why I am dividing by ten.
Next bit of magic ([H|_]) is needed, because io_lib:format() returns a LIST of strings, not a single string. So I am taking the first and single element from its top.
Last statement before end means: I am constructing a binary (<<>>), inserting {«value:», inserting the string from io_lib:format converted into binary (bytes) and closing it up with «}».
The value of the last statement is what gets returned from the function. With this fun you will get exactly { «value»: 181818 } JSON for your Adafruit feed.
Let me know if it helps.
If you will get failures, or it will not work as expected, I would like you to send me the format of the data your mote sends and the fun() you wrote for it, so I check for any errors and fix them for you.

With best regards,
Alexander Tishchenko

Viswanath95, 2017-10-13:
Hi altishchenko,
From Lorawan server i want to send a text payload to Adafruit IO[eg: Hello world]
[image]
[image]

And also in Node Configuration page I set my group name test and Application to backend there.
Still now i didn't receive any data in Adafruit IO
Thanks Alexander Tishchenko I kindly need your suggesstion

altishchenko, 2017-10-13:
Well yes, you won’t get anything :)
You just copied my example function without trying to do the one you need for your data. Say, presuming that your device sends just a text string, your function should look something like this:

fun (_Port, Text) ->
<<"{ \"value\": \"", Text/bytes, "\"}">>
end.

Give this one a try. What it does - it receives some text from your device, packs it into {"value": "yourtext"} and sends that to Adafruit. Mind the extra \" quotes, they are needed!

Best regards, 
Alexander Tishchenko
M: +7 (916) 704-61-57
E: atishch@gmail.com

altishchenko, 2017-10-14:
Hello @Viswanath95,
Did you manage to get the data in your Adafruit account?
If you are unsure, you can log in to your io.adafruit account and proceed to the API section - there you can check your feeds, if they are receiving any data right from their site by hand.

Also, you can: stop the server, clean all the log files out, start the server. See if your device registers and try sending a few messages from your mote. Then provide me with your debug.log file.

As a last resort, you can start wireshark with the filter set to tcp port 1883, then start your server and try sending a few messages from your mote, providing me with the resulting .pcap file.

As a matter of fact, did you use your adafruit _Key_ for MQTT device authentication or your own password?

altishchenko, 2017-10-15:
Hello, does everything work for you now? Can we close the issue?

Viswanath95, 2017-10-16:
Hi altishchenko,
In my io.adafruit Account On Feeds section I manually created a text feed data and on the feeds the  data is recorded on the last feed Data
[image]
And also in my io.adafruit Account on API docs on the feeds section i got the response of my manually created Data
[image]
[image]
But i have a doubt on where to find my debug.log file and .pcap file in my Lorawan server?
I use my adafruit key for MQTT Device Authentication.

altishchenko, 2017-10-17:
Hello @Viswanath95,
Sorry for the delay. To help you out with that I need to know what operating system you are running and did you install binary distribution or built the server from github repo.

Best regards,
Alexander Tishchenko

> 16 окт. 2017 г., в 10:06, Viswanath95 <notifications@github.com> написал(а):
> 
> But i have a doubt on where to find my debug.log file and .pcap file in my Lorawan server?
> I use my adafruit key for MQTT Device Authentication.

altishchenko, 2017-10-17:
Naturally, your .log files should be in the log/ directory from which you start your server. Things depend a bit on your OS and the way you start the server.

To get a .pcap file, you need:
Unix/Linux: Prior starting the server run `sudo tcpdump -i YouNetInterface -s1512 -w mqtt.pcap tcp port 1883`
Windows: Download and install wireshark. Prior running the server, start wireshark, in 'Capture using this filter' box enter `tcp port 1883` and double click your network interface.
Start the server.
Try sending a few text messages to your adafruit feed, then either press Ctrl-C in Unix/Linux or Click the 'stop' and 'save' buttons in wireshark and send me the resulting file.

Viswanath95, 2017-10-23:
Hi altishchenko,
Sorry for late reply I'm using ubuntu 14.04.2 LTS and I installed server using binary distribution.

file:///home/ethicstech/Downloads/debug.log

file:///home/ethicstech/Downloads/mqtt.pcap
And i don't know how to send my data from Lorawan to Adafruit feed
This is my Lorawan server published and subscribe Topic:
file:///home/ethicstech/Pictures/Screenshot%20from%202017-10-23%2018:28:08.png
Devviswa is my Adafruit Account Name and Send_Text_ Data_To_Adafruit is my Lorawan server backend connector name Is this is the correct way of creating a published and subscribe Topic?

Viswanath95, 2017-10-24:
Hi altishchenko,

Sorry for late reply I'm using ubuntu 14.04.2 LTS and I installed server
using binary distribution.

And i don't know how to send my data from Lorawan to Adafruit feed

This is my Lorawan server published and subscribe Topic:

Devviswa is my Adafruit Account Name and Send_Text_ Data_To_Adafruit is my
Lorawan server backend connector name Is this is the correct way of
creating a

published and subscribe Topic?

altishchenko, 2017-10-26:
Hello,
It looks like files are not attached properly. Can you try attaching them via github web interface?

Viswanath95, 2017-10-26:
Hi altishchenko,
Actually u asked me to send debug.log file and .pcap file after try to send few messages from mote.
But still now i don't know how to send messages from lorawan to Adafruit feed.
Can u guide me in this part?

altishchenko, 2017-10-26:
Yes, I'll try to guide you. But, first of all I'd like to have a look into those two files. They came to me in your message as local links to your disk, not files attached.

This link references a file on your local machine:
file:///home/ethicstech/Downloads/mqtt.pcap <file:///home/ethicstech/Downloads/mqtt.pcap>

Alexander Tishchenko
M: +7 (916) 704-61-57
E: atishch@gmail.com

Viswanath95, 2017-10-26:
> jagaJpZM4P2elu>.

Hi altishchenko,

Viswanath95, 2017-10-26:
I send the required file to your mail.

altishchenko, 2017-10-26:
I am sorry, but you sent it via github again. Please, send it to atishch@gmail.com directly, not using any replies to the previous messages. The message I received was 'Hi altishchenko,' with the body exactly like in your comment above.

gotthardp, 2017-11-14:
I assume the problem is solved, so closing. Feel free to (re)open.

closed 2017-11-14 by gotthardp