gotthardp/lorawan-server #219: Spreading factor keeps increasing.

sokolbiba, 2017-10-12:
Hi there,
I see that there are some anomalies with the Lorawan server. I can see from the graph where the spreading factor keeps increasing all the time untill it reaches the SF12. While using and testing it with the TTN network this does not seems to be problem. The node will increase the SF only when I dissconnect my gateway from the TTN network and then the next packet that will arrive will be with SF7. Do you have any idea for that? I am using the RX2 window when I setup a new device. Another thing that took my attention is that in the graph there are some "jumps" in the SF. According to graph the same packet has been sent with two different SF. Take a look at the picture below.
[image]

gotthardp, 2017-10-13:
The picture illustrates the first or the second problem or both? The device increases SF when it misses an ACK, so the Frame 12 in your picture was re-transmitted with lower rate. I guess the device has a very bad antenna, or is very far from the gateway.

sokolbiba, 2017-10-13:
The picture ilustrates both problems. I do not think that this is because of antena or the distance between the node and gateway since while I was testing in on ttn network it was working very good. I was wondering if there is any misconfiguration in the server configuration. It just surprised me because the same node worked just fine with the ttn.

gotthardp, 2017-11-14:
I forgot to answer, but I'm afraid this is a device related issue. I'd need to know the reason why the device decided to decrease the SF.

closed 2018-01-05 by gotthardp