gotthardp/lorawan-server #221: server failed to startup on windows 10

stmkjp2, 2017-10-17:
Hi, 

I installed latest Erlang (OTP20) and also latest lorawan server binary release (0.4.12) and tried to start the server by executing lorawan-server.bat but it failed miserably.

I searched all over the issues section on your github repo but I coulnd't find relevant.

I don't know what i'm doing wrong and what to do for resolve this issue. 
could you help me out identifying the problem?

following is console message when I execute the batch file and I also attached crash dump file, FYI

`=CRASH REPORT==== 17-Oct-2017::15:46:38 ===
  crasher:
    initial call: lager_crash_log:init/1
    pid: <0.110.0>
    registered_name: []
    exception error: no case clause matching {ok,log}
      in function  lager_util:expand_path/1 (/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lager/src/lager_util.erl, line 503)
      in call from lager_crash_log:init/1 (/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lager/src/lager_crash_log.erl, line 74)
      in call from gen_server:init_it/2 (gen_server.erl, line 365)
      in call from gen_server:init_it/6 (gen_server.erl, line 333)
    ancestors: [lager_sup,<0.106.0>]
    message_queue_len: 0
    messages: []
    links: [<0.107.0>]
    dictionary: []
    trap_exit: false
    status: running
    heap_size: 987
    stack_size: 27
    reductions: 218
  neighbours:

=SUPERVISOR REPORT==== 17-Oct-2017::15:46:38 ===
     Supervisor: {local,lager_sup}
     Context:    start_error
     Reason:     {{case_clause,{ok,log}},
                  [{lager_util,expand_path,1,
                       [{file,
                            "/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lager/src/lager_util.erl"},
                        {line,503}]},
                   {lager_crash_log,init,1,
                       [{file,
                            "/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lager/src/lager_crash_log.erl"},
                        {line,74}]},
                   {gen_server,init_it,2,[{file,"gen_server.erl"},{line,365}]},
                   {gen_server,init_it,6,[{file,"gen_server.erl"},{line,333}]},
                   {proc_lib,init_p_do_apply,3,
                       [{file,"proc_lib.erl"},{line,247}]}]}
     Offender:   [{pid,undefined},
                  {id,lager_crash_log},
                  {mfargs,
                      {lager_crash_log,start_link,
                          ["crash.log",65536,10485760,[{hour,0}],5]}},
                  {restart_type,permanent},
                  {shutdown,5000},
                  {child_type,worker}]

=CRASH REPORT==== 17-Oct-2017::15:46:38 ===
  crasher:
    initial call: application_master:init/4
    pid: <0.105.0>
    registered_name: []
    exception exit: {bad_return,
                     {{lager_app,start,[normal,[]]},
                      {'EXIT',
                       {{badmatch,
                         {error,
                          {shutdown,
                           {failed_to_start_child,lager_crash_log,
[... 114 lines cut ...]
               [{file,
                 "/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lager/src/lager_crash_log.erl"},
                {line,74}]},
              {gen_server,init_it,2,[{file,"gen_server.erl"},{line,365}]},
              {gen_server,init_it,6,[{file,"gen_server.erl"},{line,333}]},
              {proc_lib,init_p_do_apply,3,
               [{file,"proc_lib.erl"},{line,247}]}]}}}}},
        [{lager_app,start,2,
          [{file,
            "/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lager/src/lager_app.erl"},
           {line,219}]},
         {application_master,start_it_old,4,
          [{file,"application_master.erl"},{line,273}]}]}}}}}}},
 [{lorawan_app,start,0,
   [{file,
     "/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_app.erl"},
    {line,16}]},
  {init,start_em,1,[{file,"init.erl"},{line,1085}]},
  {init,do_boot,3,[{file,"init.erl"},{line,793}]}]}
"init terminating in do_boot",{{badmatch,{error,{lager,{bad_return,{{lager_app,start,[normal,[]]},{'EXIT',{{badmatch,{error,{shutdown,{failed_to_start_child,lager_crash_log,{{case_clause,{ok,log}},[{lager_util,expand_path,1,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lager/src/lager_util.erl"},{line,503}]},{lager_crash_log,init,1,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lager/src/lager_crash_log.erl"},{line,74}]},{gen_server,init_it,2,[{file,"gen_server.erl"},{line,365}]},{gen_server,init_it,6,[{file,"gen_server.erl"},{line,333}]},{proc_lib,init_p_do_apply,3,[{file,"proc_lib.erl"},{line,247}]}]}}}}},[{lager_app,start,2,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lager/src/lager_app.erl"},{line,219}]},{application_master,start_it_old,4,[{file,"application_master.erl"},{line,273}]}]}}}}}}},[{lorawan_app,start,0,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_app.erl"},{line,16}]},{init,start_em,1,[{file,"init.erl"},{line,1085}]},{init,do_boot,3,[{file,"init.erl"},{line,793}]}]}}
init terminating in do_boot ({{badmatch,{error,{lager,{_}}}},[{lorawan_app,start,0,[{_},{_}]},{init,start_em,1,[{_},{_}]},{init,do_boot,3,[{_},{_}]}]})

Crash dump is being written to: erl_crash.dump...done`

[erl_crash.dump.txt](https://github.com/gotthardp/lorawan-server/files/1390100/erl_crash.dump.txt)

altishchenko, 2017-10-17:
By the output, it looks like there is a problem in lorawan-server.bat:
set "ERL_ARGS=-lager log_root "log""

Somehow it gets passed into erlang as log_root=log instead of log_root="log" (missing quotes). Unfortunately, I don't have a windows machine handy to check the proper way of doing things right, but until Petr has time, you may kick start the server by deleting this line and removing %ERLARGS% from the last line of the .bat file.

Regards,
Alex

gotthardp, 2017-10-17:
That seems to be the reason.

Could you please edit your  lorawan-server.bat and change
```
set "ERL_ARGS=-lager log_root "log""
```
to
```
set ERL_ARGS="-lager log_root "log""
```
Note just one quotation mark moved. It's a very minor change.

stmkjp2, 2017-10-18:
@altishchenko @gotthardp , it works like a charm now!

Thank you for your helps and thank you for the great server app!!

Best,
JP

gotthardp, 2017-10-18:
Cool. I just did commit the fix.

closed 2017-10-18 by gotthardp