gotthardp/lorawan-server #223: Moved logs to /var/log

kazi93, 2017-10-26:
@gotthardp I have installed v0.4.12 (started in background as a daemon) by unpacking /usr/lib/lorawan-server. And start the server by systemctl start lorawan-server
However, found database in /var/lib/lorawan-server but not logs which suppose to be in /var/log/lorawan-server.  Where am I lost? Do I need to uncomment "lager" in sys.config before starting the server?

gotthardp, 2017-10-26:
Failed to open log file /var/log/lorawan-server/debug.log with error permission denied

the directory lorawan-server must be owned by the lorawan user

gotthardp, 2017-10-26:
There was one step missing from the installation guide. I just fixed it.

Thanks for reporting this and sorry for the inconvenience.

closed 2017-10-26 by kazi93