gotthardp/lorawan-server #226: How can I setup about downlink for node?

nuguri618, 2017-10-30:
Hello!
I want to know about setup for downlink.

I use feather m0 for node the use microchip gateway.

Then can get data from node.. but can't send message to node.

I use websocket client which in network server, and use JSON {"data":"01"} message send.

When node use uplink then downlinks message disappear..

Is downlink message send to node with ACK?

or need to setup other thing?

gotthardp, 2017-10-30:
Hello,
to send confirmed downlink you need to set `"confirmed":true` as described [here](https://github.com/gotthardp/lorawan-server/blob/master/doc/JSON.md).

If your downlink is displayed in web-admin and then "disappears" when a device sends uplink, then I guess the downlinks may not be working at all. Do you use ABP or OTAA join? 
Please review the "No downlink frames received" troubleshooting [here](https://github.com/gotthardp/lorawan-server/blob/master/doc/Troubleshooting.md#no-downlink-frames-delivered).

nuguri618, 2017-10-31:
@gotthardp Thanks !
I use OTTA join.
I missed setup about tx power and tx gain and tx chain.
Then now I got events about downlinks_lost.

gotthardp, 2017-10-31:
So now you need to check your gateway log and device & server configuration to find out why the downlinks get lost.

gotthardp, 2017-11-14:
Assuming the problem is fixed, so closing. Feel free to reopen.

closed 2017-11-14 by gotthardp