gotthardp/lorawan-server #227: PUSH ACK response

javierpedrido, 2017-10-30:
Hi

After upgrading to the latest code, I started seeing a lot of "ack_lost" warning messages. Then I realized that my gateway was not receiving any PUSH ACK from the server. It was a timeout problem, changing the packet forwarder parameter push_timeout_ms from 100 to 300 solved the problem. Looking at https://github.com/Lora-net/packet_forwarder/blob/master/PROTOCOL.TXT, it says that the server should send the ACK before processing the packet data. In lorawan_gw_forwarder.erl handle_info, the ACK is being sent after parsing the json object data, wouldn't be better to first send the ACK and then parse the json?

Best regards,
Javier

gotthardp, 2017-10-30:
Hello. I love your bug reports: always clear and always right. :) I just fixed it. Thank you!

javierpedrido, 2017-10-30:
Thanks for the fix! :) Maybe the same thing should be done for the PULL ACK?

gotthardp, 2017-10-30:
Yeah. Could be, but pull doesn't really do much, so the difference will be minimal.

javierpedrido, 2017-10-30:
Yes, the code before the udp send is fast.

javierpedrido, 2017-10-30:
Yes, thanks, it's working fine now with a 100 ms timeout.