gotthardp/lorawan-server #229: Thingsboard integration problem

kpranga, 2017-10-31:
I'm trying to integrate the backend of server with my thingsboard server.
Here are my problems:
1. Is it possible to send data as single parameters ? (now its inside "fields") because Thingsboard dont recognize it.
2. Is it possible to set the username of MQTT handler as variable e.g. <devadr> or <deveui>?

gotthardp, 2017-10-31:
Hello. I was not aware of Thingsboard, but it looks very nice.
1. It is not possible, but can be added easily.
2. What kind of variable (and why) would you need.

kpranga, 2017-10-31:
1. Great
2. Something went wrong with variables, updated. In Thingsboard athorization to every device is by accesstoken (unique for device, but changeable). So I figured out to set it to deveui for example so it will be operated by single handler.

kazi93, 2017-11-01:
@kpranga I am working on the same. It is pretty cool. I will update too here.

gotthardp, 2017-11-01:
@kazi93, are you developing some plug-in for the server? Shall I stop and wait for you?

kazi93, 2017-11-01:
@gotthardp Not really, please continue. I am trying to investigate the possible solution / integration. Mostly I am busy with the Mongodb -erlang adapter for the moment.

altishchenko, 2017-11-02:
Hello, just in case you might be interested, publishing data into ThingsBoard works pretty good. To make it work for one device you need to set Handlers format to 'Raw Data' and provide a 'Parse Uplink' function that will build JSON with values for your device: {"temperature" : "25.6", "humidity" : "27.0", "active" : "false"} etc.Don't know how to change device validations automatically though.

altishchenko, 2017-11-02:
Have a look into Integrations.md, Adafruit section.

kpranga, 2017-11-02:
But still, I think its not a solution.  Is it efficient in terms of server memory/processor time to use for example 300 Handlers for 300devices? :) Besides of work hours to set it up.

altishchenko, 2017-11-02:
Actually, jsx: set of functions is available from Parse Uplink, so you can use this to build utf8 JSON binary for it:

```
JsonBin = jsx:encode([{<<"temperature">>, TempVal}, {<<"humidity">>, HumVal}, {<<"active">>, false}])
```
will result in:
```
JsonBin = <<"{\"temperature\":25.4,\"humidity\":60,\"active\":false}">>
```

altishchenko, 2017-11-02:
> But still, I think its not a solution.
Agree to that. As for the work hours - it can be just a script that will load handlers/connectors into mnesia db without doing it all by hand.

gotthardp, 2017-11-04:
@kpranga, the thingsboard.io talks about per-device Access Tokens or Certificates. I think that in any case there has to be one MQTT connection per device (which is not very resource friendly) and the question is only how the setup will be done, right?
You wanted to have a template-based username, so I guess you planned to change the default Access Token and use some device-specific parameter and the Token. Is this a correct assumption?

kpranga, 2017-11-09:
Yes, you are right.

Kralg, 2017-11-09:
Hello, Thingsboard looks very interesting. Can someone share experience of similar projects for custom dashboards or customize UI/singledashboard which memory/processor efficient?

altishchenko, 2017-11-09:
Hello,

I am not (yet) sure which memory/processor is adequate for your needs, but ThingsBoard looks really nice. There is a small catch in it - to have the process of building dashboards and everything streamlined, you have to start with the device publishing some JSON to v1/me/telemetry, then you can shortcut through the menus, otherwise you do have to go all the way of creating a device, a dashboard for it, and then all the widgets you may want to get from it. Have a look at https://thingsboard.io/docs/getting-started-guides/helloworld/ - it should give you some ideas. In case of configuring MQTT, I'd suggest you go through Integrations doc (Adafruit topic) and their link to MQTT <https://thingsboard.io/docs/reference/mqtt-api/#key-value-format> guide page.

And be warned, ThingsBoard authenticates on a per-device basis! So, if you have multiple devices - you must have multiple connectors. OR, do it as I do it - one connector, but different JSON variable names. Like, instead of just "temperature" - "temperature-102-seiling" - then you can use one connector for every device.

With best regards,
Alexander Tishchenko
M: +7 (916) 704-61-57
E: atishch@gmail.com

Kralg, 2017-11-10:
Hi,

Thanks a lot. Yes, I do have multiple devices, and I stated thinking what would happen if the number of them grow to several hundreds or more. This is why I started thinking of something efficient. Also I think that authentication on a per device and MQTT connector for each of the device is madness in my case. However single  connector sounds much better.

kpranga, 2017-11-27:
Hello,
@altishchenko Do you know how to access devaddr when passing uplink? I want to temporary use your method with differenet JSON variable names like temperature-devaddr(variable) : value
Thanks in advance

altishchenko, 2017-11-27:
Well, in publish topic you can use {devaddr} macro expansion. But, ask Petr
if it is at all possible to get devaddr in the uplink fun. (I think not).

gotthardp, 2017-11-27:
The "macros" work in the topic only. For now. In the 0.5.0 I want to make it more consistent, so it should be possible in the future.

gotthardp, 2018-01-12:
Hi there. The ThingsBoard integration is now possible. It will be included in the future 0.5.0.

closed 2018-01-12 by gotthardp

cristianhumelnicu, 2018-03-23:
Hi Petr,

Could you tell me what is supouse to to for integrate Loraserver with TheThinkboard?
Thank you

altishchenko, 2018-03-24:
Hello Cristian (@cristianhumelnicu ) did you have a look at Integration.md document? There is a section on ThingsBoard.

cristianhumelnicu, 2018-03-25:
Hi @altishchenko Yes but i can not yet figure out what is the best metod to enrole and to publish in thethingboard 1000 devices 

Could you guide me a little bit ? 

Thx!

altishchenko, 2018-03-26:
To the server or thingsboard? For the server side, you can do this:
- create one device
- do a dbexport (the file you need is nodes.js)
- generate your devices by some script in the format of nodes.js
- dbimport nodes.js back (just delete all other .js in the export directory)

altishchenko, 2018-03-26:
sorry, nodes.json not .js

cristianhumelnicu, 2018-03-26:
HI @altishchenko 

Is all about thingsboard platform , how to transmit data from gotthard server ... 
As you know thingsboard request token authentification for each device ...

So if I have 1000 lora device who transmit data from gotthard lora server how i will integrate easy in thinkboard? Each of loradevice transmit 8 parameters  decripted with cayenne lpp connector from gotthard loraserver.. and i want to integrate it in thingsboard 

I can manage all functions from loraserver ... 

Than you!

gotthardp, 2018-03-28:
@cristianhumelnicu, please try to follow the Handlers.md, Connectors.md and (mainly) Integration.md documentation and open a new issue when you get stuck.

cristianhumelnicu, 2018-04-25:
@gotthardp , Hi Petr, ive done all things from integration.md for connecting to thingsboard.io

Could you tell me if will working  with cayenne lpp decoder or is need it to parse and build a json like @altishchenko has describe?

gotthardp, 2018-04-25:
If your device sends the LPP payload, the using LPP decode should work.

cristianhumelnicu, 2018-04-25:
@gotthardp , Yes is send LPP but i wanna plublish in thingsboard.io and i've seen in debug. log connector is failed to connect to "mqtt://demo.thingsboard.io" 

2018-04-25 13:00:58.443 [debug] <0.725.0>@lorawan_connector_mqtt:handle_info:230 Connector <<"TB">> to <<"mqtt://demo.thingsboard.io">> (attempt31) failed: {shutdown,{connack_error,'CONNACK_AUTH'}} (count: 33)

cristianhumelnicu, 2018-04-25:
That mean ....  client is not authorized to connect ...

gotthardp, 2018-04-25:
That is correct. Fix the user credentials.

cristianhumelnicu, 2018-04-25:
Credentials is corect .... I just try to logout and lon in again to demo.thingsboar.io with same user and paswd from connector

cristianhumelnicu, 2018-04-25:
Setting for connector and handler is down
[image]
[image]

cristianhumelnicu, 2018-04-25:
And here connector authentification ... i did remove password from this picture
[image]

cristianhumelnicu, 2018-04-25:
And i have this from debug.log

2018-04-25 15:33:18.580 [warning] <0.1228.0>@lorawan_connector_mqtt:terminate:247 Connector TB terminated: {{shutdown,nxdomain},{gen_fsm,sync_send_event,[<0.1650.0>,{<0.1228.0>,ping},10000]}}
2018-04-25 15:33:18.581 [error] <0.1228.0> gen_server <0.1228.0> terminated with reason: {{shutdown,nxdomain},{gen_fsm,sync_send_event,[<0.1650.0>,{<0.1228.0>,ping},10000]}} in gen_fsm:sync_send_event/3 line 252
2018-04-25 15:33:18.581 [error] <0.1228.0> CRASH REPORT Process <0.1228.0> with 0 neighbours exited with reason: {{shutdown,nxdomain},{gen_fsm,sync_send_event,[<0.1650.0>,{<0.1228.0>,ping},10000]}} in gen_fsm:sync_send_event/3 line 252
2018-04-25 15:33:18.581 [error] <0.294.0> Supervisor lorawan_connector_sup had child {mqtt,<<"TB">>} started with lorawan_connector_mqtt:start_link({connector,<<"TB">>,<<"Cayenne">>,<<"json">>,<<"mqtt://demo.thingsboard.io">>,<<"v1/devices/me/tel...">>,...}) at <0.1228.0> exit with reason {{shutdown,nxdomain},{gen_fsm,sync_send_event,[<0.1650.0>,{<0.1228.0>,ping},10000]}} in context child_terminated
2018-04-25 15:33:18.582 [debug] <0.1653.0>@lorawan_connector_mqtt:connection_args:123 Connecting TB to <<"mqtt://demo.thingsboard.io">>, id <<"F2kWJg61A66MBDEpIo73">>, user <<"cristian.humelnicu@gmail.com">>
2018-04-25 15:33:18.582 [debug] <0.294.0> Supervisor lorawan_connector_sup started lorawan_connector_mqtt:start_link({connector,<<"TB">>,<<"Cayenne">>,<<"json">>,<<"mqtt://demo.thingsboard.io">>,<<"v1/devices/me/tel...">>,...}) at pid <0.1653.0>
2018-04-25 15:33:18.588 [debug] <0.1653.0>@lorawan_connector_mqtt:handle_info:230 Connector <<"TB">> to <<"mqtt://demo.thingsboard.io">> (attempt311) failed: {shutdown,nxdomain} (count: 0)

gotthardp, 2018-04-25:
`nxdomain` suggests you may have problems with domain resolution

cristianhumelnicu, 2018-04-25:
I run lora server from windows ... Maybe is related with that ?!

cristianhumelnicu, 2018-04-25:
it is handler and connector settings ok ? I've post 3 pictures upper

cristianhumelnicu, 2018-04-25:
More from log.......

2018-04-25 15:57:57.202 [info] <0.2455.0> device 001B9EBFDE75DF7E {join,<<"062BE26D">>}
2018-04-25 15:57:57.202 [debug] <0.2455.0>@lorawan_mac:encode_accept:393 Join-Accept <<"062BE26D">>, netid <<1,2,3>>, cflist undefined, rx1droff 0, rx2dr 5, appkey <<"5B5C69A08357963048FEB3F0DC08C907">>, appnce <<"F9510E">>
2018-04-25 15:57:57.202 [debug] <0.2455.0>@lorawan_handler:join:107 Join-Accept in RX1: {rxq,868.1,<<"SF7BW125">>,<<"4/5">>,{{2018,4,25},{12,57,56.826343}},244519134,-39,7.0}
2018-04-25 15:57:57.222 [warning] <0.290.0>@lorawan_gw_forwarder:handle_info:115 Weird data <<2,83,183,4,36,10,196,255,254,0,242,66,123,34,116,120,112,107,95,97,99,107,34,58,32,123,34,101,114,114,111,114,34,58,32,34,78,79,78,69,34,125,125>>
2018-04-25 15:57:58.652 [warning] <0.289.0> gateway 240AC4FFFE00F242 {downlinks_lost,1}
2018-04-25 15:58:03.217 [debug] <0.1742.0>@lorawan_connector_mqtt:reconnect:136 Reconnecting
2018-04-25 15:58:03.348 [warning] <0.1742.0>@lorawan_connector_mqtt:terminate:247 Connector TB terminated: {{badmatch,{error,waiting_for_connack}},[{lorawan_connector_mqtt,'-handle_info/2-fun-6-',1,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_connector_mqtt.erl"},{line,223}]},{lists,foreach,2,[{file,"lists.erl"},{line,1338}]},{lorawan_connector_mqtt,handle_info,2,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_connector_mqtt.erl"},{line,221}]},{gen_server,try_dispatch,4,[{file,"gen_server.erl"},{line,616}]},{gen_server,handle_msg,6,[{file,"gen_server.erl"},{line,686}]},{proc_lib,init_p_do_apply,3,[{file,"proc_lib.erl"},{line,247}]}]}
2018-04-25 15:58:03.348 [error] <0.1742.0>@lorawan_connector_mqtt:-handle_info/2-fun-6-:223 gen_server <0.1742.0> terminated with reason: no match of right hand value {error,waiting_for_connack} in lorawan_connector_mqtt:'-handle_info/2-fun-6-'/1 line 223
2018-04-25 15:58:03.349 [error] <0.1742.0>@lorawan_connector_mqtt:-handle_info/2-fun-6-:223 CRASH REPORT Process <0.1742.0> with 0 neighbours crashed with reason: no match of right hand value {error,waiting_for_connack} in lorawan_connector_mqtt:'-handle_info/2-fun-6-'/1 line 223
2018-04-25 15:58:03.349 [error] <0.2460.0>@lorawan_connector_mqtt:-handle_info/2-fun-6-:223 gen_fsm <0.2460.0> in state waiting_for_connack terminated with reason: no match of right hand value {error,waiting_for_connack} in lorawan_connector_mqtt:'-handle_info/2-fun-6-'/1 line 223
2018-04-25 15:58:03.349 [error] <0.2460.0>@lorawan_connector_mqtt:-handle_info/2-fun-6-:223 CRASH REPORT Process <0.2460.0> with 1 neighbours exited with reason: no match of right hand value {error,waiting_for_connack} in lorawan_connector_mqtt:'-handle_info/2-fun-6-'/1 line 223 in gen_fsm:terminate/7 line 602
2018-04-25 15:58:03.349 [debug] <0.2462.0>@lorawan_connector_mqtt:connection_args:123 Connecting TB to <<"mqtt://demo.thingsboard.io">>, id <<"F2kWJg61A66MBDEpIo73">>, user <<"cristian.humelnicu@gmail.com">>
2018-04-25 15:58:03.349 [error] <0.294.0>@lorawan_connector_mqtt:-handle_info/2-fun-6-:223 Supervisor lorawan_connector_sup had child {mqtt,<<"TB">>} started with lorawan_connector_mqtt:start_link({connector,<<"TB">>,<<"Cayenne">>,<<"json">>,<<"mqtt://demo.thingsboard.io">>,<<"v1/devices/me/tel...">>,...}) at <0.1742.0> exit with reason no match of right hand value {error,waiting_for_connack} in lorawan_connector_mqtt:'-handle_info/2-fun-6-'/1 line 223 in context child_terminated
2018-04-25 15:58:03.349 [debug] <0.294.0> Supervisor lorawan_connector_sup started lorawan_connector_mqtt:start_link({connector,<<"TB">>,<<"Cayenne">>,<<"json">>,<<"mqtt://demo.thingsboard.io">>,<<"v1/devices/me/tel...">>,...}) at pid <0.2462.0>
2018-04-25 15:58:03.622 [debug] <0.2462.0>@lorawan_connector_mqtt:handle_info:230 Connector <<"TB">> to <<"mqtt://demo.thingsboard.io">> (attempt311) failed: {shutdown,{connack_error,'CONNACK_AUTH'}} (count: 0)
2018-04-25 15:58:03.622 [debug] <0.2462.0>@lorawan_connector_mqtt:reconnect:136 Reconnecting
2018-04-25 15:58:03.884 [debug] <0.2462.0>@lorawan_connector_mqtt:handle_info:230 Connector <<"TB">> to <<"mqtt://demo.thingsboard.io">> (attempt31) failed: {shutdown,{connack_error,'CONNACK_AUTH'}} (count: 1)
2018-04-25 15:58:08.837 [debug] <0.2462.0>@lorawan_connector_mqtt:reconnect:136 Reconnecting
2018-04-25 15:58:09.100 [debug] <0.2462.0>@lorawan_connector_mqtt:handle_info:230 Connector <<"TB">> to <<"mqtt://demo.thingsboard.io">> (attempt311) failed: {shutdown,{connack_error,'CONNACK_AUTH'}} (count: 2)
2018-04-25 15:58:19.013 [debug] <0.2462.0>@lorawan_connector_mqtt:reconnect:136 Reconnecting
2018-04-25 15:58:19.276 [debug] <0.2462.0>@lorawan_connector_mqtt:handle_info:230 Connector <<"TB">> to <<"mqtt://demo.thingsboard.io">> (attempt31) failed: {shutdown,{connack_error,'CONNACK_AUTH'}} (count: 3)

altishchenko, 2018-04-26:
@cristianhumelnicu What do you use for password with thingsboard.io? Your own password or 'device token'? If I remember correctly, for each device reporting to thingsboard you should use a device token specific to that device.

cristianhumelnicu, 2018-04-26:
@gotthardp , @altishchenko  Hi Petr, Alex .
I did add  to connector the communication port 8883 , try 1883 and did not working, so is   mqtt://demo.thingsboard.io:8883 For credential i've use cristian.humelnicu@gmail and password from platform and working is same way . I mean seems is authentificate but when sensor transmit in platform the connector in crusshing  shuting down. I run gotthard server on windows so maybe is related with that. Did not try yet on linux or thingsboard on premises ...

This is part of debug.log
018-04-26 00:06:44.100 [debug] <0.295.0>@lorawan_connector_sup:start_child:16 Start {mqtt,<<"ThingsBoard">>}
2018-04-26 00:06:44.104 [debug] <0.1109.0>@lorawan_connector_mqtt:connection_args:123 Connecting ThingsBoard to <<"mqtt://demo.thingsboard.io:8883">>, id <<"pSwkW6wBTqlDym8Uj5Y5">>, user <<"cristian.humelnicu@gmail.com">>
2018-04-26 00:06:44.104 [debug] <0.294.0> Supervisor lorawan_connector_sup started lorawan_connector_mqtt:start_link({connector,<<"ThingsBoard">>,<<"CayenneAPP">>,<<"json">>,<<"mqtt://demo.thingsboard.io:8883">>,<<"...">>,...}) at pid <0.1109.0>
2018-04-26 00:07:05.134 [debug] <0.1109.0>@lorawan_connector_mqtt:handle_info:230 Connector <<"ThingsBoard">> to <<"mqtt://demo.thingsboard.io:8883">> (attempt311) failed: {shutdown,etimedout} (count: 0)
2018-04-26 00:07:05.134 [debug] <0.1109.0>@lorawan_connector_mqtt:reconnect:136 Reconnecting
2018-04-26 00:07:14.576 [debug] <0.1109.0>@lorawan_connector_mqtt:handle_info:198 WTF {<<"v1/devices/me/attributes">>,[]}
2018-04-26 00:07:14.781 [info] <0.1116.0> device 001B9EBFDE75DF7E {join,<<"074CFEA2">>}
2018-04-26 00:07:14.781 [warning] <0.1116.0> node 074CFEA2 {repeated_reset,1}
2018-04-26 00:07:14.781 [debug] <0.1116.0>@lorawan_mac:encode_accept:393 Join-Accept <<"074CFEA2">>, netid <<1,2,3>>, cflist undefined, rx1droff 0, rx2dr 5, appkey <<"5B5C69A08357963048FEB3F0DC08C907">>, appnce <<"401DB4">>
2018-04-26 00:07:14.781 [debug] <0.1116.0>@lorawan_handler:join:110 Join-Accept in RX2: {rxq,868.1,<<"SF7BW125">>,<<"4/5">>,{{2018,4,25},{21,7,13.914982}},2525487141,-37,6.0} {0,5,869.525}
2018-04-26 00:07:14.793 [warning] <0.290.0>@lorawan_gw_forwarder:handle_info:115 Weird data <<2,214,229,4,36,10,196,255,254,0,242,66,123,34,116,120,112,107,95,97,99,107,34,58,32,123,34,101,114,114,111,114,34,58,32,34,78,79,78,69,34,125,125>>
2018-04-26 00:07:16.254 [warning] <0.289.0> gateway 240AC4FFFE00F242 {downlinks_lost,1}
2018-04-26 00:07:26.144 [debug] <0.1109.0>@lorawan_connector_mqtt:handle_info:230 Connector <<"ThingsBoard">> to <<"mqtt://demo.thingsboard.io:8883">> (attempt31) failed: {shutdown,etimedout} (count: 1)
2018-04-26 00:07:31.147 [debug] <0.1109.0>@lorawan_connector_mqtt:reconnect:136 Reconnecting
2018-04-26 00:07:46.713 [info] <0.1118.0> device 001B9EBFDE75DF7E {join,<<"074CFEA2">>}
2018-04-26 00:07:46.713 [warning] <0.1118.0> node 074CFEA2 {repeated_reset,2}
2018-04-26 00:07:46.713 [debug] <0.1118.0>@lorawan_mac:encode_accept:393 Join-Accept <<"074CFEA2">>, netid <<1,2,3>>, cflist undefined, rx1droff 0, rx2dr 5, appkey <<"5B5C69A08357963048FEB3F0DC08C907">>, appnce <<"997012">>
2018-04-26 00:07:46.713 [debug] <0.1118.0>@lorawan_handler:join:107 Join-Accept in RX1: {rxq,868.1,<<"SF7BW125">>,<<"4/5">>,{{2018,4,25},{21,7,45.844971}},2557417125,-39,7.0}
2018-04-26 00:07:46.723 [warning] <0.290.0>@lorawan_gw_forwarder:handle_info:115 Weird data <<2,226,123,4,36,10,196,255,254,0,242,66,123,34,116,120,112,107,95,97,99,107,34,58,32,123,34,101,114,114,111,114,34,58,32,34,78,79,78,69,34,125,125>>
2018-04-26 00:07:52.162 [warning] <0.1109.0>@lorawan_connector_mqtt:terminate:247 Connector ThingsBoard terminated: {{shutdown,etimedout},{gen_fsm,sync_send_event,[<0.1117.0>,{<0.1109.0>,ping},10000]}}
2018-04-26 00:07:52.162 [error] <0.1109.0> gen_server <0.1109.0> terminated with reason: {{shutdown,etimedout},{gen_fsm,sync_send_event,[<0.1117.0>,{<0.1109.0>,ping},10000]}} in gen_fsm:sync_send_event/3 line 252
2018-04-26 00:07:52.192 [error] <0.1109.0> CRASH REPORT Process <0.1109.0> with 0 neighbours exited with reason: {{shutdown,etimedout},{gen_fsm,sync_send_event,[<0.1117.0>,{<0.1109.0>,ping},10000]}} in gen_fsm:sync_send_event/3 line 252
2018-04-26 00:07:52.192 [debug] <0.1119.0>@lorawan_connector_mqtt:connection_args:123 Connecting ThingsBoard to <<"mqtt://demo.thingsboard.io:8883">>, id <<"pSwkW6wBTqlDym8Uj5Y5">>, user <<"cristian.humelnicu@gmail.com">>
2018-04-26 00:07:52.192 [error] <0.294.0> Supervisor lorawan_connector_sup had child {mqtt,<<"ThingsBoard">>} started with lorawan_connector_mqtt:start_link({connector,<<"ThingsBoard">>,<<"CayenneAPP">>,<<"json">>,<<"mqtt://demo.thingsboard.io:8883">>,<<"...">>,...}) at <0.1109.0> exit with reason {{shutdown,etimedout},{gen_fsm,sync_send_event,[<0.1117.0>,{<0.1109.0>,ping},10000]}} in context child_terminated
2018-04-26 00:07:52.192 [debug] <0.294.0> Supervisor lorawan_connector_sup started lorawan_connector_mqtt:start_link({connector,<<"ThingsBoard">>,<<"CayenneAPP">>,<<"json">>,<<"mqtt://demo.thingsboard.io:8883">>,<<"...">>,...}) at pid <0.1119.0>
2018-04-26 00:08:13.206 [debug] <0.1119.0>@lorawan_connector_mqtt:handle_info:230 Connector <<"ThingsBoard">> to <<"mqtt://demo.thingsboard.io:8883">> (attempt311) failed: {shutdown,etimedout} (count: 0)
2018-04-26 00:08:13.206 [debug] <0.1119.0>@lorawan_connector_mqtt:reconnect:136 Reconnecting
[... cut ...] ontext child_terminated
2018-04-26 00:09:00.247 [debug] <0.294.0> Supervisor lorawan_connector_sup started lorawan_connector_mqtt:start_link({connector,<<"ThingsBoard">>,<<"CayenneAPP">>,<<"json">>,<<"mqtt://demo.thingsboard.io:8883">>,<<"...">>,...}) at pid <0.1145.0>
2018-04-26 00:09:11.383 [debug] <0.295.0>@lorawan_connector_sup:stop_child:23 Stop {mqtt,<<"ThingsBoard">>}
2018-04-26 00:09:11.383 [debug] <0.1145.0>@lorawan_connector_mqtt:terminate:244 Connector ThingsBoard terminated: shutdown

cristianhumelnicu, 2018-04-26:
BTW i've try user: cristian.humelnicu@gmail.com and password {token } and did not connected in anyway

altishchenko, 2018-04-26:
@cristianhumelnicu I am extremely sorry, I just reviewed my old connectors a little too quickly. If you had a look at the link I provided, you could have noticed, that device **access token** should be used as a username, not password - this one should be either left blank (try it out) or should be yours. As I remember, I left it blank in the past and it worked. Second, the data format should be JSON, with fields appropriately denoted, like: {"temperature": 21.3, "humidity": 75}. I've never heard that thingsboard understands other fancy formats, like LPP.

cristianhumelnicu, 2018-04-26:
@altishchenko  Thank you for info Alex. That mean the is not working like Petr  has describe in integration.md document. I just folow the steps from there and seems is not working ,,, 
@gotthardp Do u have any ideea about it ?

gotthardp, 2018-04-26:
@cristianhumelnicu, any idea about what?

altishchenko, 2018-04-26:
@cristianhumelnicu What you should do is: 1. Use JSON; 2. Use your device access token as a **username**.

cristianhumelnicu, 2018-04-27:
@altishchenko  Ok Alex , i will tryto do that  and if do not work I will open one new issue. 

Thank you !

cristianhumelnicu, 2018-04-28:
Hi Petr , Hi Alex 

Is working with {apparg} at name and nothing to pass or other field.... Also i can confirm is working with Cayenne LPP too if node sending in this format and thingsboard platform will recive field1 , field2 .. etc with values of that fields ..

Thank you @gotthardp , @altishchenko !