gotthardp/lorawan-server #230: "Confirm" shows crossed (x) in "Received Frames" v.0.4.12

kazi93, 2017-11-03:
@gotthardp Hi! Greetings! Could you please confirm what am I missing. Why does it shows cross icon in the  Confirm attribute or feature. 
[image]

gotthardp, 2017-11-03:
Hello. It's means the message is "unconfirmed", i.e. not confirmed (see LoRaWAN message types).

kazi93, 2017-11-03:
@gotthardp So, created handlers, connectors and parsing as the same way I did with v 0.4.10. Currently running two server with v0.4.10 and v0.4.12 with the same config and same sensor. So, I do not really understand why is it showing crossed and unknown_appid.

gotthardp, 2017-11-03:
The crossed symbol is unrelated to unknown_appid.
 * It is crossed because it was sent as unconfirmed; it's not an error. (Earlier versions had a bug, which showed a tick instead a cross).
 * The unknown app is displayed when the Node's *Group* name does not correspond to any handler. Could there be a typo?

kazi93, 2017-11-04:
It is not a typo. However, it resolved. Thank you

closed 2017-11-04 by kazi93

gotthardp, 2017-11-04:
Was it due to a bug in the server or documentation?

susiloharjo, 2018-05-24:
how to resolved this @kazi93 ?