gotthardp/lorawan-server #231: lorawan_mac_region:datars(undefined)

altishchenko, 2017-11-03:
Hello Petr,
Started getting these errors since last upgrade from very long ago version. Have any idea what is causing this?
Here is a bit from debug.log:
2017-11-03 10:14:34.516 [warning] <0.8368.0> server {http_error,{404,"/devstat/4
C1D448E"}}
2017-11-03 10:14:34.518 [error] <0.202.0> Lager event handler error_logger_lager
_h exited with reason {'EXIT',{{case_clause,[http,<0.8364.0>,4,<0.8374.0>,functi
on_clause,[{lorawan_mac_region,datars,[undefined],[{file,"/home/netcube/lorawan-
server/_build/default/lib/lorawan_server/src/lorawan_mac_region.erl"},{line,132}
]},{lorawan_mac_region,datar_to_dr,2,[{file,"/home/netcube/lorawan-server/_build
/default/lib/lorawan_server/src/lorawan_mac_region.erl"},{line,167}]},{lorawan_a
dmin_rxgraph,'-get_rxframe/2-fun-0-',1,[{file,"/home/netcube/lorawan-server/_bui
ld/default/lib/lorawan_server/src/lo..."},...]},...]]},...}}
2017-11-03 10:14:34.526 [debug] <0.202.0>@lager_handler_watcher:119 Lager instal
led handler error_logger_lager_h into error_logger
2017-11-03 10:14:37.298 [debug] <0.8155.0>@ciscoms_apci:process_apdu:174 recv: S
-ACK(154)
2017-11-03 10:14:39.521 [error] <0.8378.0>@lorawan_mac_region:datars:132 CRASH R
EPORT Process <0.8378.0> with 0 neighbours crashed with reason: no function clau
se matching lorawan_mac_region:datars(undefined) line 132

gotthardp, 2017-11-03:
The 'undefined' parameter shall be a Region. Did you set it for all your devices/nodes?

altishchenko, 2017-11-03:
Surely! It looks somehow web-admin triggered. Were there any changes in mnesia tables from .10 or .11 ? It does not happen, if I don't touch Status tabs.
Here is a bit from crash.log:
2017-11-03 10:14:34 =CRASH REPORT====
  crasher:
    initial call: cowboy_stream_h:request_process/3
    pid: <0.8374.0>
    registered_name: []
    exception error: {function_clause,[{lorawan_mac_region,datars,[undefined],[{
file,"/home/netcube/lorawan-server/_build/default/lib/lorawan_server/src/lorawan
_mac_region.erl"},{line,132}]},{lorawan_mac_region,datar_to_dr,2,[{file,"/home/n
etcube/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_mac_region.e
rl"},{line,167}]},{lorawan_admin_rxgraph,'-get_rxframe/2-fun-0-',1,[{file,"/home
/netcube/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_admin_rxgr
aph.erl"},{line,59}]},{lists,map,2,[{file,"lists.erl"},{line,1239}]},{lists,map,
2,[{file,"lists.erl"},{line,1239}]},{lorawan_admin_rxgraph,get_rxframe,2,[{file,
"/home/netcube/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_admi
n_rxgraph.erl"},{line,55}]},{cowboy_rest,call,3,[{file,"/home/netcube/lorawan-se
rver/_build/default/lib/cowboy/src/cowboy_rest.erl"},{line,1128}]},{cowboy_rest,
set_resp_body,2,[{file,"/home/netcube/lorawan-server/_build/default/lib/cowboy/s
rc/cowboy_rest.erl"},{line,1019}]}]}
    ancestors: [<0.8364.0>,<0.359.0>,<0.358.0>,ranch_sup,<0.238.0>]
    message_queue_len: 0
    messages: []
    links: [<0.8364.0>]
    dictionary: []
    trap_exit: false
    status: running
    heap_size: 6772
    stack_size: 27
    reductions: 5093
  neighbours:

gotthardp, 2017-11-03:
True. Probably some old record in DB. I made a fix that could solve this. Will you try that, please?

altishchenko, 2017-11-03:
Definitely. Let me fetch/merge.

gotthardp, 2017-11-03:
No, wait :)

gotthardp, 2017-11-03:
Now. Please fetch and try :)

altishchenko, 2017-11-03:
Hmm, looks like still there. Want a full debug.log+crash.log?
I'll try to attach them:
[crash.log](https://github.com/gotthardp/lorawan-server/files/1441556/crash.log)
[debug.log](https://github.com/gotthardp/lorawan-server/files/1441562/debug.log)

gotthardp, 2017-11-03:
Yes, because the error was there twice and I fixed just one occurrence. (I was too fast, sorry.)
Please try again.

altishchenko, 2017-11-03:
Doing it.

altishchenko, 2017-11-03:
While I am compiling, this to tends to travel between releases for a while:
src/lorawan_admin_devstat.erl:64: Warning: variable 'Batt' is unused
src/lorawan_admin_devstat.erl:64: Warning: variable 'Margin' is unused
src/lorawan_admin_devstat.erl:64: Warning: variable 'Timestamp' is unused

altishchenko, 2017-11-03:
Only lorawan_admin_devstat:encode_timestamp(undefined) is left now.

gotthardp, 2017-11-03:
Ok, fixed also.

altishchenko, 2017-11-03:
All gone. Thanks a lot!

closed 2017-11-03 by altishchenko