gotthardp/lorawan-server #233: Garbages displayed in the beginning of some received data in "Received Frames" page

dynaxis, 2017-11-10:
I use 0.4.13 and though I slightly modified some of regional parameters, I believe it's not related to this issue.

As shown in the picture below, some data starts with garbages like: `.\.%,......|.#.........y3v.....D...1......)NCG`, which is unexpected since the data is supposed to be in hexdecimal digits. Actually the expected data doesn't seem overwritten. They follow the garbages. Also when I extract the same data via websocket, I see no such garbages there.

I copy a couple of such packets here for your reference. Payload size is 110 bytes:
Via the received frames page:
```.\.%,......|.#.........y3v.....D...1......)NCG..a...r'>02002B00E0F68BCA285A20403AF37EA26F9BCE921EFB1AC827590EA0FD8A14FDDAF20C3337D3E123B106FADA9584994EEB8128D55D71D9783EDE5CCC252CFBFA90ADED087CDB23C32E199FE4EF92D7E3793376A19D0BAEC344FB1A183182C9A0ED9301294E43471E1861FCB2C172```

Via websocket:
```02002B00E0F68BCA285A20403AF37EA26F9BCE921EFB1AC827590EA0FD8A14FDDAF20C3337D3E123B106FADA9584994EEB8128D55D71D9783EDE5CCC252CFBFA90ADED087CDB23C32E199FE4EF92D7E3793376A19D0BAEC344FB1A183182C9A0ED9301294E43471E1861FCB2C172```

Via the received frames page:
```....;.Mj.........d..O.|.?T..$:.;.....k'>02002700E5B885F1067CFBFB0FB25342106B611B3CB345489CF5E4E36D8DF99601EF546573962522F51D5F64DA579D8802862ABCE06A0E5DAD6F2952656FC1F24409B962871BC83EFE12FCFB3B154D6AFC0CBEE7B8C0FC1FC16489074F9A7CEC3F54BDF4243AAB3B91B499DA086B```

Via websocket:
```02002700E5B885F1067CFBFB0FB25342106B611B3CB345489CF5E4E36D8DF99601EF546573962522F51D5F64DA579D8802862ABCE06A0E5DAD6F2952656FC1F24409B962871BC83EFE12FCFB3B154D6AFC0CBEE7B8C0FC1FC16489074F9A7CEC3F54BDF4243AAB3B91B499DA086B```

[image]

gotthardp, 2017-11-10:
I believe this may be because the message contained the `>` character, which prematurely terminated the [ASCII] string which should display as a hint.

gotthardp, 2017-11-10:
I just made a fix that might solve this problem (see above). You can apply this patch without rebuilding the binaries as it's in the Javascript only.

dynaxis, 2017-11-11:
Thanks. I'll test and get back to you with the result.

dynaxis, 2017-11-11:
I tested with the current `master` branch and got no such garbages any more.
Thanks for your effort.

closed 2017-11-11 by dynaxis