gotthardp/lorawan-server #235: How to post JSON data to restful web services

h2onet, 2017-11-15:
Hello, 

I'm using the semtech mote configuration and I'm receiving correct data from the remote Lora board in both OTAA and ABP modes.
Now, I would like to send those data to a rest service on the same machine as lorawan-server in json format: 
HTTP POST method, 
URI: http://localhost:8081/LoraService/api/sensors, 
no authentication, 
but I'm not sure about the backend configuration. 
Following the instructions from the page: https://github.com/gotthardp/lorawan-server/blob/master/doc/Backends.md I have created a new Handler:
Group: Sensors (the same name as for the lora device in OTAA mode) 
Format: JSON
Uplink Fields: empty
Parse Uplink: fun(_Port, <<LED, Press:16, Temp:16, AltBar:16, Batt, Lat:24, Lon:24, AltGps:16>>) ->#{led => LED, pressure => Press, temp => Temp/100, alt_bar => AltBar, batt => Batt}
end.
Build Downlink: empty
Connector: JSONTest

and a new CONNECTOR:
Connector Name: JSONTest
Enabled: On
URI: http://localhost:8081/LoraService/api/sensors
Published Topic: empty
All others fields are empty, also in the  authentication tab.
Is this configuration correct? Unfortunately, there is no data at the rest server side. 
Thanks,

gotthardp, 2017-11-15:
First of all, are you sure your device sends data in the format `<<LED, Press:16, Temp:16, AltBar:16, Batt, Lat:24, Lon:24, AltGps:16>>`?
Have a look into debug.log to see if there are some errors or warnings.

h2onet, 2017-11-15:
Hello,

I use the B-L072Z-LRWAN1 board from STMicro with the EndNode application from I-CUBE-LRWAN software expansion. The code seems the same as on the reference page https://github.com/Lora-net/LoRaMac-node/blob/master/src/apps/LoRaMac/classA/LoRaMote/main.c#L207. 
The frame from the lorawan-server Received Frames window: 00000008700000FE3E090D0503AB0000  Led status: 0,  temperature: 2160, battery: 254, pressure and altitude: 0. 
I would like to send those data to the rest server. 
I have moved the restful service to the new address  http://192.168.1.64:8080/LoraService/api/sensors to avoid any conflict. But so far with no results. I have attached the last part of the debug log.
Thanks,
Maczek

[debug.log](https://github.com/gotthardp/lorawan-server/files/1476204/debug.log)

gotthardp, 2017-11-15:
Are you using server v0.4.13 (the latest)? It looks like the bug I fixed 14 days ago.

matiasz8, 2017-11-15:
Hello @gotthardp ,
I'm a little confuse about this too. 
I'm using server v0.4.13, the last version, which one have in Handlers section Uplink Fields. 
I'm using a Semtech-mote for the region US902-928, working with ABP mode.
Actually I can send packets from the Semtech-mote and view as listed in "Received Frames". 
I need to send a message to Semtech-mote Class C from my application.

Can I use the Loraserver Rest Api in this way?
Thank,
Nahuel

gotthardp, 2017-11-15:
Hello @matiasz8, downlink (sending data to a device) via HTTP is not yet supported. Will be in 0.5.0. Meanwhile you can use websockets or MQTT.

h2onet, 2017-11-16:
@gotthardp Now, I have the latest version v0.4.13 up and running (upgrade from v0.5.12). Unfortunately, there is no change. I have recorded the network trafic on the port 8080 and there are no packets with the request POST going out. In the Received Frames window I can see the correct data from the Lora board.
Maczek

[debug2.log](https://github.com/gotthardp/lorawan-server/files/1478193/debug2.log)

gotthardp, 2017-11-16:
@h2onet The **application** is set to **semtech-mote**. You should set it to **Backend**.

matiasz8, 2017-11-16:
Thanks @gotthardp for the answer. i'll check it out,
Nahuel

h2onet, 2017-11-17:
@gotthardp I changed the application to "Backend". But it did not help. The rest service is up and running and reacheble for the lorawan-server. There is a new debug log.
Maczek
[debug3.log](https://github.com/gotthardp/lorawan-server/files/1482052/debug3.log)

gotthardp, 2017-11-17:
The system tries to establish the HTTP connection, but it fails for some reason. Could double-check the HTTP URI and Authorization and (if it does not help) try to inspect the HTTP traffic with Wireshark, please?

h2onet, 2017-11-20:
@gotthardp The request fails because the server is trying to send data to http://192.168.1.64:8080 and not to the URI defined in Connectors: http://192.168.1.64:8080/LoraService/api/sensors
Maczek

altishchenko, 2017-11-20:
@h2onet Hello. Can you try moving the URI-path stuff to the Published Topic section? Leave the URI box with just http://192.168.1.64:8080 and in the box below write /LoraService/api/sensors . I think this will do the trick for you.

h2onet, 2017-11-20:
@altishchenko Hello. I already did it. But then I'm getting a new error message "HTTP Request Failed: 400"

h2onet, 2017-11-20:
@altishchenko and @altishchenko thank you for helping out. Now, it's working fine. @altishchenko I followed your advice. In fact the error was to put in the Published Topic field (Connectors tab) LoraService/api/sensors instead of /LoraService/api/sensors (leading forward slash problem). I suggest to update Backend (Connectors section) documentation on that point. Maybe it will be also helpful to override the semtech-mote option with Backend for the devices when defining a Handler for a group.

The working version for the rest services (simple POST Method - without authentication):
1. Set application to Backend.
2. Handler configuration:
Group: Sensors (the same name as for the lora device in OTAA mode)
Format: JSON
Uplink Fields: empty
Parse Uplink: fun(_Port, <<LED, Press:16, Temp:16, AltBar:16, Batt, Lat:24, Lon:24, AltGps:16>>) ->#{led => LED, pressure => Press, temp => Temp/100, alt_bar => AltBar, batt => Batt}
end.
Build Downlink: empty
Connector: JSONTest
3. Connector configuration:
Connector Name: JSONTest
Enabled: On
URI: http://192.168.1.64:8080
**Published Topic: /LoraService/api/sensors**

altishchenko, 2017-11-20:
Great it works for you! We will definitely do. And as far as I know, semtech-mote may not survive transition to the 0.5.0. Configuration will change to be more intuitive and not much cumbersome like it is now.

gotthardp, 2017-11-20:
I will keep this open until we fix the documentation.