gotthardp/lorawan-server #236: AWS certificate upload fails with 405

balkat, 2017-11-16:
Version 4.1.3
When configuring an AWS IoT connector the upload of the certificate and private key fails.
Debug log shows:

2017-11-16 07:11:58.296 [warning] <0.11630.0> server {http_error,{405,"/admin/upload"}}

Have tried with firefox and chrome -- same result.

gotthardp, 2017-11-16:
It works for me. Could you please have a look in the debug.log if there is some warning or an error message.

balkat, 2017-11-17:
Hi Peter,

Have attached a debug log from yesterday.  Were a lot of http errors in it.
This morning the upload worked fine.   Possible conjecture is the other http errors
temporarily caused the upload to fail (despite having restarted the service a couple of times).  Having done nothing overnight its worked on the first attempt today....

If I can isolate a scenario that causes the same problem I will let you know.  For now, all is OK.

[debug.log.0.txt](https://github.com/gotthardp/lorawan-server/files/1480756/debug.log.0.txt)

altishchenko, 2017-11-17:
Browser cache from the previous server interactions, quite possible?

balkat, 2017-11-18:
I recall the http errors started with the google maps part when setting up
a new gateway.

balkat, 2017-11-18:
Also, I did try two separate browsers, and a server restart, with no
improvement.

gotthardp, 2017-11-18:
Would you be able to capture the HTTP traffic with Wireshark? I need more information for debugging.

altishchenko, 2017-11-18:
Hmm, server restart will not affect browser caches. There may be a catch with the `gun` module, because it tries to keep the connection open all of the time. If your server serves the request and right after this closes the connection, it is possible to get these errors. BUT, according to your own debug output - this is not the case. It looks like your HTTP server is just rejecting the requests in the first place. Obviously, you can try it out by hand - telnet to the port of your application, type in all necessary fields and data, and have a look at what's sent in reply.

Example:
```
telnet localhost 8080
GET / HTTP/1.1
Host: localhost

HTTP/1.1 200 Ok
...
```

balkat, 2017-11-19:
I am travelling to a conference and will be away until early December.  Will see if I can replicate the problem when I get home.

waynestallwood-kcom, 2017-12-22:
I also have this issue and am unable to upload my certs. I am running from the docker image (tag latest)

Tried on Safari and Chrome (with a clear cache)

2017-12-22 11:25:11.564 [warning] <0.6015.0> server {http_error,{405,"/admin/upload"}}
2017-12-22 11:25:19.461 [warning] <0.6111.0> server {http_error,{405,"/admin/upload"}}

waynestallwood-kcom, 2017-12-22:
I've created a wireshark dump of one upload failing, see here
https://s3-eu-west-1.amazonaws.com/ws-share/lorawan-certupload.pcapng

Is there another way to install the certs, whilst this issue is resolved ?..It's a bit of a blocker for me now as everything else is working perfectly apart from the AWS integration.

gotthardp, 2017-12-22:
There seems to be some problem with the content-type.

Would you be able to send HTTP PUT with a JSON payload? The easiest way around is to:
 * upload the file manually into `/var/lib/lorawan-server`, or try some command line tool to upload the file
 * HTTP GET /connectors/AWS%20IOT and store the data to some file
 * manually edit the file and modify the `certfile` attribute to include the certificate filename (no path, just the name)
 * HTTP PUT /connectors/AWS%20IOT the modified file back

gotthardp, 2017-12-22:
On Docker the `/var/lib/lorawan-server` is the shared directory where is the `Mnesia...` and the `log` directory and the `.erlang.cookie` file.

waynestallwood-kcom, 2017-12-22:
Thanks for this, I am struggling with the HTTP GET. 

curl http://admin:mypass@127.0.0.1:8080/admin/#/connectors/AWS%20IOT returns page source and 
url http://admin:admin@127.0.0.1:8080/connectors/AWS%20IOT returns nothing. I suspect I'm not making it past the authentication.

gotthardp, 2017-12-22:
I think you should switch the authentication
``curl --digest --user name:password http://127.0.0.1:8080/connectors/AWS%20IOT``

waynestallwood-kcom, 2017-12-22:
Ok that's worked THANK YOU ! for your support, I'm now pushing data into AWS IoT. For the sake of others and until the upload bug is resolved, here are the curl commands I used.

curl --digest --user admin:mypass http://127.0.0.1:8080/connectors/AWS%20IOT >> dump.json

Edit the file for key names after uploading them

curl --digest -H "Content-Type: application/json" --user admin:mypass http://127.0.0.1:8080/connectors/AWS%20IOT --upload-file dump.json

gotthardp, 2017-12-22:
Cool! Thanks a lot for your cooperation.

I will keep this bug open until the bug is fixed.

gotthardp, 2018-01-21:
i hope this is also fixed in 0.5.0

closed 2018-01-21 by gotthardp