gotthardp/lorawan-server #237: Improve ISSUE template

gotthardp, 2017-11-17:
I'd suggest adding stuff like:
For the better understanding of the issue you are encouraged to provide:

 -   Distribution type: binary or built from source.
 -   Hardware type: PC Server, Just PC, MacBook, MacServer, Raspberry Pi1/2/3/x, Odroid, etc.
  -  OS type and version. Also, add the output of the uname -a command if you are running on a sort of a Unix/Linux/MacOS system.
  -  Your Erlang/OTP version. Can be obtained by running erl at the command line: all the text before 1> line.
  -  If you are experiencing problems with the web-interface or seeing lots of http_error lines in the log , provide npm -v and node -v output too.
 -   Get ready to provide your server's debug.log file and, in hard cases, the Wireshark network capture file.

altishchenko, 2017-11-17:
Actually, the last point shan't be on the list :) It should go as an achtung a bit lower :)

gotthardp, 2018-01-12:
Documented for 0.5.0.
I put the text into the Troubleshooting guide to keep the ISSUE_TEMPLATE very short. People are not reading long texts.

closed 2018-01-12 by gotthardp